fkie/catkin_lint #19: Spurious unknown dependencies in some environments

roehling, 2015-01-07:
This is to be investigated: ros/rosdistro#6519

roehling, 2015-01-08:
@wjwwood: if `ROS_DISTRO` is unset, rosdep won't return any catkin packages at all. As `catkin_lint` searches `ROS_PACKAGE_PATH` as well, this does not hurt most of the times, but it can lead to complains about unknown dependencies for packages which exist but are not installed (in my case: `ros-indigo-map-server`).

As you can see, I added a sanity check for `catkin` itself. There is little more I can do from my end.

wjwwood, 2015-01-08:
@roehling I'm not sure this is the best heuristic. How are you asking rosdep for resolutions? Are you calling the rosdep CLI or using the rosdep API?

Ideally the user would be notified that the ROS distribution could not be determined and then either ask the user to specify the correct one or assume a default ROS distribution. Whether or not the ROS_DISTRO environment variable is set is a pretty reliable way to check this.

Many of our tools require the user to specify the ROS distro on the command line if the ROS_DISTRO environment variable is not set.

Thanks for taking the time to look into it.

roehling, 2015-01-08:
I am using the rosdep API here: https://github.com/fkie/catkin_lint/blob/master/src/catkin_lint/linter.py#L447
Any suggestions on how to improve that would be appreciated.

wjwwood, 2015-01-08:
Ok, I'll have a look at that and let you know if I come up with a better way to validate dependencies.

roehling, 2015-01-14:
I changed the logic to mimic the behavior of the rosdep CLI, i.e. the new --rosdistro option overrides the ROS_DISTRO environment variable. If neither is set, catkin_lint complains and ignores all unknown dependencies. I also revamped the rosdep database queries. Unfortunately, I cannot reliably distinguish catkin packages from other packages using rosdep. I did find the "_is_ros" key, but that is set to true for cmake build type packages as well.

wjwwood, 2015-01-14:
Ok cool, sorry I haven't had time to look into it yet, but you seem to be a step a head of me :smile:.

> Unfortunately, I cannot reliably distinguish catkin packages from other packages using rosdep. I did find the "_is_ros" key, but that is set to true for cmake build type packages as well.

This is where the `rosdistro` and the `rosdistro` cache would be useful. You could use the `rosdistro` python library to see if a rosdep is a released package and if it is you can get the `package.xml` from the cache, which would let you check to see what the `build_type` is (`cmake` versus `catkin`).

So basically a rosdep will be in one of three (and a half) categories:

- It is a rosdep key which maps to a system dependency
- It is a released ROS package
 - In this case you can check the `package.xml` for the `build_type`, `catkin` or `cmake` or something else
- It is either an invalid rosdep key (like a typo or something) or it is an unreleased package.

If it falls into the last category you can check the local system (RPP) to see if the user has it from source.

Unfortunately there isn't much in the way of documentation for the `rosdistro` python library, but you can have a look at some examples of how we use it:

- https://github.com/ros-infrastructure/rosinstall_generator
- https://github.com/ros-infrastructure/ros.org_scripts/blob/master/wiki.ros.org/indigo_metapackage_status.py
- https://github.com/ros-infrastructure/bloom/blob/master/bloom/commands/release.py#L171-274

Hopefully that gives you some more insight into the issue.

Thanks again for working on this, it's something I would have liked to do, given the time.

roehling, 2015-01-15:
Thank you for this extremely helpful hint. I already have a caching mechanism for local package.xml files, so hopefully I can replace or at least enhance it with `rosdistro`.

roehling, 2015-01-15:
I looked at the examples, and as far as I can tell, the initial `get_index()` call always triggers a remote download, which is kind of undesirable (not to mention slow in our company network). Is there a way to use `rosdistro` without an active internet connection?

wjwwood, 2015-01-15:
The `rosdistro` python library always downloads the index in order to get the latest one. If you wanted to cache the index you could using pickle or caching the actual source yaml file. I don't have an example of this, but it should be possible.

If you get a cached distribution (different from the index) then it will also download a cache file which is a tar.gz of all of the package.xml's which will be even slower. However, there really isn't an alternative for some tools, like `rosinstall_generator`. We implement local caching with this library, but it would be possible, the API is pretty modular.

roehling, 2015-01-16:
@wjwwood I have come up with a potential solution. Would you mind testing it against your use cases and see if `catkin_lint` behaves satisfactorily?

wjwwood, 2015-01-16:
That looks pretty good, the only comment I have is that you can optionally download all package manifests at once rather than downloading them one by one by using `rosdistro.get_cached_distribution`. That is almost certainly faster than getting them one by one. That also makes the cache more useful run to run.

wjwwood, 2015-01-16:
The other point is that I see you've set the cache invalidation to 30 days, which is a fine default, but there needs to be away to tell it to download new information right now, the rosdistro changes daily.

roehling, 2015-01-23:
I decided against `get_cached_distribution` for two reasons: 

First, there seems to be no canonical per-user cache location like `rosdep` has it, so I would end up storing the cache in a `catkin_lint` specific location and be unable to take advantage of pre-cached data from other programs.

Second, as I understand the code, it merely downloads the package manifests one by one, which I can do myself more efficiently by limiting the download to those packages which I actually need (I only use `rosdistro` as a last resort if no local information is available, as I feel that the locally installed packages should override the official distribution sources anyway).

Regarding the cache invalidation, I will implement an `--update-rosdistro` just to be safe, although I don't expect it will be needed very often: the `rosdistro` index itself is not cached and will be fetched each time `catkin_lint` wants to download data, so the cache merely prevents `catkin_lint` from learning about changes in previously downloaded manifests.

wjwwood, 2015-01-23:
> First, there seems to be no canonical per-user cache location like rosdep has it, so I would end up storing the cache in a catkin_lint specific location and be unable to take advantage of pre-cached data from other programs.

Perhaps I am misunderstanding it, but isn't that already the case with the files you are downloading individually? AFAICT they all go into a `catkin_lint` specific cache and aren't shared across applications.

> Second, as I understand the code, it merely downloads the package manifests one by one, which I can do myself more efficiently by limiting the download to those packages which I actually need (I [...]

This is not actually the case. We explicitly added this to make it more efficient to get many manifest all at once. The build farm compiles a tar.gz of all package manifests once every 5 minutes.

If you only need one or two package manifests it will be on the same order of magnitude as the single cache, but I would encourage you to try the two methods on a package with many dependencies.

`rosinstall_generator`, for instance, downloads this cache every single time (never caches) and it takes less then 5 seconds on my machine (I used `rviz` as an example). However, if I run `catkin_lint` on something like `rviz` where it has to download many package manifests for its dependencies it takes much longer. I ran it on `rviz` after clearing the cache and it took 55 seconds.

> Regarding the cache invalidation, I will implement an --update-rosdistro just to be safe, although I don't expect it will be needed very often: the rosdistro index itself is not cached and will be [...]

I agree that some cache is probably appropriate, but it's always good to make people aware of the caching and give them an easy option for refreshing it.

Maybe my version is out-of-date but for me it doesn't seem to re-download the index each time.

I believe that using the `rosdistro.get_cached_distribution` method and caching that result would be better than fetching them individually, but I understand that refactoring to do that will take some time and that you've already invested quite a lot in this particular issue.

roehling, 2015-01-23:
> I believe that using the `rosdistro.get_cached_distribution` method and caching that result would be better than fetching them individually, but I understand that refactoring to do that will take some time and that you've already invested quite a lot in this particular issue.

I don't mind changing the code if it improves performance. Is `rosdistro_build_cache` using a different mechanism than that? Because I tried that and it was very slow on my system, which led me to believe fetching package manifests individually is faster.

wjwwood, 2015-01-23:
`rosdistro_build_cache` is what the farm uses to generate the cache which is downloaded by the `rosdistro.get_cached_distribution` function. It would be slow since it downloads all manifests for all released packages.

roehling, 2015-01-27:
The refactoring turned out to be really simple: `s/get_distribution/get_cached_distribution/`.
For the rosdistro update, the `--clear-cache` option (which was previously undocumented) will invalidate all cache entries, so `catkin_lint` will re-download missing package manifests and re-parse all local package.xml files.

wjwwood, 2015-01-27:
That all seems great. I'll give it a spin later today. Thanks again for working on this!

roehling, 2015-02-11:
I assume everything is okay and close the issue.

closed 2015-02-11 by roehling