fkie/catkin_lint #22: "error: unconfigured build_depend" for Python package message dependenciess

jack-oquin, 2015-08-21:
Of course, this error message can be avoided by declaring only an ``<exec_depend>`` or ``<run_depend>``, but that is contrary to recommend usage:

 * http://docs.ros.org/api/catkin/html/howto/format2/python_module_dependencies.html
 * http://docs.ros.org/api/catkin/html/howto/format1/python_module_dependencies.html

The consensus of the ROS buildsystem SIG was that message packages should always have build and run or exec dependencies, even when not strictly necessary, as is the case for pure Python packages.

I don't feel really strongly about this, but I recently found myself adding unnecessary ``find_package()`` and ``catkin_package()`` CMakeLists.txt configuration commands to suppress that error, so I thought it worth mentioning.

I have no good idea how to tell message dependencies from others, and this error message clearly does much more good than harm, in general. Maybe if a package calls ``catkin_python_setup()`` and has no ``add_executable()`` or ``add_library()`` it can be assumed to be purely Python and hence not in need of CMake build dependency configuration.

If this issue does not seem worth the effort, go ahead and close it.

Many thanks for your work on catkin_lint! I continually reap significant benefits from it.

roehling, 2015-09-09:
I decided to go with the least intrusive change and demote the diagnostic to warning level if neither `add_executable()` nor `add_library()` has been used. If that check has no unintended side effects, we can make it a notice or drop it altogether at a later date.

jack-oquin, 2015-09-09:
That sounds good. We certainly don't want it to miss any genuine omissions.

roehling, 2016-01-10:
Any update on this issue? Shall we leave it as a warning or make it a notice?

jack-oquin, 2016-01-13:
I defer to your judgement, Timo. I don't fully understand the various options or their risks and benefits.

roehling, 2016-02-15:
I decided to leave it as it is for now, and address any complaints or issues if they come up.

closed 2016-02-15 by roehling