fkie/catkin_lint #23: suggestion: make "target 'foo' is not installed" a warning, not a notice

jack-oquin, 2015-08-21:
I frequently ignore most of the other "notices", but this one is important when preparing a package for release.

jack-oquin, 2015-08-21:
Looking at the code, the same point could be made for UNINSTALLED_EXPORT_LIB and maybe MISSING_INSTALL_INCLUDE. (I am unsure what [info.commands](https://github.com/fkie/catkin_lint/blob/master/src/catkin_lint/checks/build.py#L358) actually contains.

I'll be happy to provide a pull request, if you agree.

roehling, 2015-09-09:
`info.commands` is the list of all commands in the `CMakeLists.txt`. Basically, if there is no `install()` command at all, I assume that people don't care about proper installation, since most packages work just fine from develspace. But I agree that this diagnostic is important for package releases.

closed 2015-09-09 by roehling

jack-oquin, 2015-09-09:
Looks great. Thanks!

commit bd2fb0c4 (2015-09-09): Make uninstalled targets a warning-level diagnostic

Closes #23