fkie/catkin_lint #24: Erroneous Error on Exec Depends That are Also Test Depends

evenator, 2015-08-31:
I have found that if I find_package() a dependency for unit tests (in my case, roslaunch), catkin_lint will find an error unless the dependency is listed as a test_depend or a build_depend. Listing the package as an exec_depend does not fulfill the requirement for catkin_lint.

According to http://wiki.ros.org/catkin/package.xml, "Test Dependencies specify only additional dependencies for unit tests. They should never duplicate any dependencies already mentioned as build or run dependencies. " It seems that specifying the package as an exec_depend fulfills this requirement, so catkin_lint shouldn't consider this an error.

frederikhegger, 2015-09-08:
Additionally, when using the roslaunch file check, another error is reporting on "duplicate find_package(catkin)". The CMakeLists looks like the following: 

```
cmake_minimum_required(VERSION 2.8.3)
project(mir_grasp_monitors)

find_package(catkin REQUIRED
  COMPONENTS
    roscpp
    std_msgs
)

catkin_package(
  CATKIN_DEPENDS
    rospy
    sensor_msgs
    std_msgs
)

....

### TESTS
if(CATKIN_ENABLE_TESTING)
  find_package(catkin REQUIRED
    COMPONENTS
      roslaunch
  )

  roslaunch_add_file_check(ros/launch)
endif()
```

roehling, 2015-09-09:
Actually, the latter is a legitimate error, because the second `find_package(catkin)` call will not work properly thanks to the CMake internal caching mechanism. You should use `find_package(roslaunch)` instead. I'm working on the fix for the original problem.

frederikhegger, 2015-09-09:
Okay, thanks for the hint.

commit dade9b99 (2015-09-09): Run dependencies do not need to be test dependencies

Closes #24
closed 2015-09-09 by roehling