fkie/catkin_lint #25: errors from unresolved environment variables

T045T, 2015-10-07:
I've just run into a problem where we use environment variables in our include paths, like so:
```
include_directories(
  ${catkin_INCLUDE_DIRS}
  $ENV{dependency_ROOT}/include
  )
```

This leads to the following error:
```
error: build include path '$ENV{dependency_ROOT}/include' does not exist
```
Now, I'm aware the use of environment variables is strongly discouraged, but still, the path pointed to by this *does* exist.

I'm ready to put together a Pull Request that implements environment variable expansion, but I wanted to discuss if and where to make this change.

To me it seems like there are two obvious places to resolve environment variables: ```real_path()``` (in linter.py) and ```_resolve_vars()``` (in cmake.py).

I think changing ```_resolve_vars()``` (or adding an equivalent ```_resolve_env_vars()``` function makes the most sense.
Changing the behavior of ```_resolve_vars()``` would change the outcome of ```test_substitution()```, like so:
```
self.assertEqual(
    self.parse_all("cmd($ENV{PATH})"),
    [ ("cmd", [ "$ENV{PATH}" ]) ]
)
```
would become
```
self.assertEqual(
    self.parse_all("cmd($ENV{PATH})"),
    [ ("cmd", [ os.path.expandvars('${PATH}' ]) ]
)
```

One argument for changing ```real_path()``` instead is that linter.py already imports os and uses it heavily, whereas cmake.py does not.

What do you think?

roehling, 2015-10-07:
That change should probably go into real_path(), or even better, make it a dedicated resolve_env_vars() method in the linter class that can be called by real_path().

closed 2015-10-08 by roehling

roehling, 2015-12-10:
Just to let you know: I released a new version (1.4.4), but I changed the logic a little bit: the default behavior is like before your pull request, but you can use the command line option "--resolve-env" to resolve $ENV{} references.