fkie/catkin_lint #27: Check alphabetical order for listed components?

wkentaro, 2015-11-06:
I mean

* packages names in run_depend and build_depend
* file names for msg and srv

wkentaro, 2015-11-06:
I think it should be ordered in alphabetical order, because people can duplicate and mis some dependencies or files if there are lots of things.

roehling, 2015-11-09:
For message and service files, `catkin_int` already issues a notice (`-W2`) if the lists are not sorted. Since `catkin_lint` relies on `catkin_pkg` for parsing the `package.xml`, it cannot check the order in that file. It does, however, check for missing dependencies, and I could add a check for duplicate entries.

wkentaro, 2015-11-09:
> For message and service files, catkin_int already issues a notice (-W2) if the lists are not sorted. Since catkin_lint relies on catkin_pkg for parsing the package.xml

Thanks, I found that `catkin_lint` has that feature already after creating this issue.

closed 2015-12-10 by roehling