fkie/catkin_lint #29: Kinetic version of catkin_lint?

jack-oquin, 2016-06-09:
I recently installed Xenial and ROS Kinetic, but ``sudo apt-get install python-catkin-lint`` does not work.

There are other ways to install it, but I wonder if there is a plan to release it to the Kinetic repositories?

roehling, 2016-06-10:
Hm, there is a Xenial version in my Launchpad PPA. Maybe @tfoote can help?

jack-oquin, 2016-06-10:
Your PPA is fine for now. I didn't think of looking there.

I guess the advantage of adding it to the ROS repository is for packages to list it as a dependency.

tfoote, 2016-06-10:
I imported the new ppa build into the ROS repos. The packages will be available shortly.

roehling, 2016-06-12:
Thanks, @tfoote.

I think you can remove quantal, raring, and saucy from the list since they reached EOL and have been removed from the PPA. Furthermore, I started packaging a Python 3 version which you may want to import as well (`python3-catkin-lint`).

tfoote, 2016-06-13:
Imported, and they'll propogate with the next syncs

closed 2016-06-13 by roehling