fkie/catkin_lint #30: endif(expression) a notice?

v4hn, 2016-06-20:
Heya,

I just hit the pretty obscure notice "notice: extra arguments in endif()".
Why does catkin_lint complain about that - even if it's just a notice?
It is standard practice to add the same expression as in the if-block there, if the block
exceeds a certain length. This is to make it more easy to understand the logical structure
of the build script.

In my opinion this notice should be removed again, as it's just noise among more helpful notices like
"list ... should be sorted" or "operands for operator VERSION_LESS should be quoted strings" (same project...)

cmake's help states:
> Note that the expression in the else and endif clause is optional.

This is something entirely different from "the use is discouraged" (which is what a lint notice usually implies)

roehling, 2016-06-21:
The [CMake coding standards](http://docs.ros.org/api/catkin/html/user_guide/standards.html) for catkin state that closing keywords like `endif()` and `endforeach()` should not repeat the condition.

In other words, it's a matter of policy (just like indentation) and happens to be decided for ROS in this particular way. If it bothers you, you can either use the `--ignore endblock_args` command line option or add a comment saying `#catkin_lint: ignore endblock_args` near the beginning (before the first `if()`) of your `CMakeLists.txt`.

v4hn, 2016-06-21:
Thanks for the reference! I didn't know of that point in the style guide.
So now I'll keep wondering why someone decided on that peculiar policy for
endif blocks (why not leave that choice to the user...) instead of blaming
catkin_lint for noticing it. :-)

roehling, 2016-06-21:
Personally, I think the ROS style guide is good the way it is. For instance, an `else()` statement with condition looks almost the same as an `elseif()`, but has the opposite effect.  That being said, if you really need the condition repeated and do not want to be irritated by `catkin_lint` notices, why not just add a comment like this:

    if(some-condition)
        ...
    else() # some-condition
        ...
    endif() # some-condition

v4hn, 2016-06-21:
On Tue, Jun 21, 2016 at 06:15:53AM -0700, Timo Röhling wrote:
> if(some-condition)
>     ...
> else() # some-condition
>     ...
> endif() # some-condition

That's a nice workaround! Thanks!

closed 2016-06-21 by roehling