fkie/catkin_lint #36: Debian packaging

roehling, 2017-03-10:
* [Official Introduction to Debian packaging](https://wiki.debian.org/Packaging/Intro?action=show&redirect=IntroDebianPackaging)
* [Tutorial for GBP](http://honk.sigxcpu.org/projects/git-buildpackage/manual-html/gbp.html)
* [Debian Policy Manual](https://www.debian.org/doc/debian-policy/)
* [Debian Python Policy](https://www.debian.org/doc/packaging-manuals/python-policy/)
* [Python Library Style Guide](https://wiki.debian.org/Python/LibraryStyleGuide)

roehling, 2017-03-10:
@j-rivero I opened a new issue for the Debian packaging discussion. 

I think I'm good on the technical side. I am maintaining a dedicated packaging branch `debian` already, using `pybuild` to build the package. Of course, feel free to point out any problems you notice.

I would like some input in policy matters, though. `catkin_lint` is primarily a command-line tool with a two modules, `catkin_lint` and `catkin_lint.checks`. The modules basically implement the tool itself, so could be considered private. On the other hand, there is a public API which allows people to extend `catkin_lint` with their own tests. This would make part of the modules public.

Right now, everything is installed as public modules. I could also abandon the (probably never used) extension feature and make everything private. Or, I could separate out the public parts into its own dedicated module. Any recommendations?

Also, I am currently building two packages for Python 2 and Python 3 respectively, but they cannot be installed together because they both install the same binary and manpage. Does it even make sense to have both packages or should I just build the package once for whatever the default Python version is on the target distribution?

rctoris, 2018-07-11:
Is there any update on going forward with this? ROS Melodic now builds releases on Debian (stretch to be specific) and we depend on catkin_lint for https://github.com/fetchrobotics/fetch_tools so this the lack of this is preventing us from releasing.

roehling, 2018-07-12:
I still need a [sponsor](https://wiki.debian.org/Glossary#sponsor) to upload the package. As soon as I find some Debian member willing to upload the package on my behalf, I will proceed. Unfortunately, I don't know any.

j-rivero, 2018-07-12:
Ouch sorry for not answering before I just saw this thread from the last updates. I can help reviewing the debian packaging and even maintain the package. For the upload, we can ask @jspricke to be our sponsor when we found some time.

jspricke, 2018-07-12:
Awesome, I can sponsor you, sure :). I would propose to drop the python2 version, unless you know of any users. Also, you need to file a ITP in the Debian issue tracker, clean up the changelog...
Ping me if you want a complete review.

Also, I would propose to move the packaging branch into the debian-science group on salsa.debian.org, that's where all ROS packages in Debian are hosted.

roehling, 2018-08-31:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=907144

The package has been uploaded to Debian unstable. Many thanks to @jspricke for his support and sponsorship!

closed 2018-08-31 by roehling