fkie/catkin_lint #38: Support for catkin_install_python

dseifert, 2017-05-11:
Per http://docs.ros.org/kinetic/api/catkin/html/howto/format2/installing_python.html, one should use `catkin_install_python` to install python scripts. However, catkin_lint does not recognize this command and consequently warns about uninstalled files.

Example CMakeLists.txt:

```
cmake_minimum_required(VERSION 2.8.3)
project(my_test_project)

find_package(catkin REQUIRED)

catkin_package(
	CATKIN_DEPENDS
	DEPENDS
)

install(
	DIRECTORY launch
	DESTINATION
		${CATKIN_PACKAGE_SHARE_DESTINATION}
)

catkin_install_python(PROGRAMS
	scripts/${PROJECT_NAME}
	DESTINATION ${CATKIN_PACKAGE_SHARE_DESTINATION}/scripts
)
```

where `catkin lint -W2` complains about `my_test_project: warning: executable file 'scripts/my_test_project' is not installed` even though it is.

closed 2017-05-11 by roehling

dseifert, 2017-05-11:
I'm sorry to say that this doesn't work. It handles `install_catkin_python`, however the command is `catkin_install_python`. If I correct this in build.py, it works fine.

roehling, 2017-05-12:
Stupid typo :-(

Fixed it

dseifert, 2017-05-12:
Thanks, this works nicely.

commit a487da44 (2017-05-11): Add support for install_catkin_python() command

Fixes #38