fkie/catkin_lint #40: False positive for exporting plugins to base classes of the same commit

ipa-mdl, 2017-10-17:
`catkin_lint` reports an error if no dependences on the the plugins' base classes were listed, which is great.
However, this leads to false postives if the base class is part of the exporting packages and suggests the packages to depend on their selves.

Examples from  `ros_canopen`:
```
canopen_402: warning: package exports canopen_402 plugin but does not exec_depend on 'canopen_402'
     *         Your package exports a plugin for another package, but
     * fails to list said         package as a dependency.
     * You can ignore this problem with --ignore plugin_depend
canopen_master: warning: package exports canopen_master plugin but does not exec_depend on 'canopen_master'
socketcan_interface: warning: package exports socketcan_interface plugin but does not exec_depend on 'socketcan_interface' 
```

ipa-mdl, 2017-10-18:
duplicate of #39, my bad :-/
Works with current master, sorry for the noise!

closed 2017-10-18 by ipa-mdl

roehling, 2017-10-25:
This reminds me that I probably should release a new version soon.