fkie/catkin_lint #41: Dynamic reconfigure .cfg files are executable

argenos, 2017-10-21:
When running catkin_lint --strict dynamic reconfigure [cfg files](http://wiki.ros.org/dynamic_reconfigure/Tutorials/HowToWriteYourFirstCfgFile) throw the uninstalled_script warning since they are executable, but are not properly installs. 
They are only used at build time as far as I understand, but they need to be executable.

roehling, 2017-10-26:
`catkin_lint` has a special rule to ignore executable files in the `cfg` subfolder, which AFAIK is the recommended file system location for `dynamic_reconfigure` configuration files.

argenos, 2017-10-27:
Thanks for your answer! Is there a way to handle other directories? The repository I work on prefers `config` vs `cfg` and we have a couple dozen cfg files that are causing trouble.

roehling, 2017-10-28:
Now that I think of it, I should be able to enumerate all config files from `CMakeLists.txt`. I will add the pertinent code to `catkin_lint` once I'm back from vacation.

roehling, 2017-11-06:
Fixed in 1.4.17

closed 2017-11-06 by roehling

argenos, 2017-11-28:
Hi @roehling , do you know when release 1.4.17 will be available in apt?