fkie/catkin_lint #45: internal error: 'Package' object has no attribute 'version' abi

alex-gee, 2018-03-05:
If I run `catkin_lint <pkg_name>` on any package I get a
>catkin_lint: internal error: 'Package' object has no attribute 'version_abi'

Running `catkin_lint <pkg_name> --debug` yields

> catkin_lint: internal error: 'Package' object has no attribute 'version_abi'
>
>Traceback (most recent call last):
>  File "/usr/local/bin/catkin_lint", line 31, in <module>
>    main()
>  File "/usr/local/lib/python2.7/dist-packages/catkin_lint/main.py", line 188, in main
>    sys.exit(run_linter(args))
>  File "/usr/local/lib/python2.7/dist-packages/catkin_lint/main.py", line 109, in run_linter
>    env.add_path(path)
>  File "/usr/local/lib/python2.7/dist-packages/catkin_lint/environment.py", line 150, in add_path
>    pkgs = find_packages(path, use_cache=self.use_cache)
>  File "/usr/local/lib/python2.7/dist-packages/catkin_lint/environment.py", line 68, in find_packages
>    _load_cache()
>  File "/usr/local/lib/python2.7/dist-packages/catkin_lint/environment.py", line 244, in _load_cache
>    _cache = pickle.loads(f.read())
>AttributeError: 'Package' object has no attribute 'version_abi'

A current workaround is running with `--disable-cache`. Any ideas where this is comming from?

roehling, 2018-03-05:
Yes, this is a known issue in the caching code (see #44). The underlying problem has been fixed since version 1.4.18; unfortunately, it has not been synched with the official ROS repository yet (@tfoote ?).

You can fix the problem with older versions by running  `catkin_lint --clear-cache` once.

alex-gee, 2018-03-05:
Ok thanks for the quick answer.

closed 2018-03-05 by alex-gee

roehling, 2018-03-08:
Thanks, I'll keep that in mind!