fkie/catkin_lint #46: CMake Subdirectory Handling Broken

cellard0or, 2018-03-20:
Commit https://github.com/fkie/catkin_lint/commit/a0dfd73ce52cdc87cda01dbee1ad35ae44d3d5e6 introduced a change which makes the linter fail because of wrongly concatenated file paths. Specifically the two lines here  https://github.com/fkie/catkin_lint/commit/a0dfd73ce52cdc87cda01dbee1ad35ae44d3d5e6#diff-6d6ab9f4d4f345d92797be8db11d669fR216.

Consider the minimal project:
[catkin_lint_test.zip](https://github.com/fkie/catkin_lint/files/1829606/catkin_lint_test.zip)

When running `catkin_lint` in the current version inside it, the error
> catkin_lint_test: tests/secondLevel/CMakeLists.txt(1): error: add_library() needs missing file 'tests/tests/secondLevel/Test.cpp'
occurs. 
The reason is that in the new version `info.subdir` now contains the path twice.

Reverting these two lines fixes the problem for me but I am not deep enough into the code to propose a complete patch at this point.

I hope we can fix this together :)

roehling, 2018-03-20:
You are correct, the line you mention introduced a regression.

closed 2018-03-20 by roehling

cellard0or, 2018-03-20:
Wow, that was fast. Works fine, thanks a lot!

roehling, 2018-03-20:
You're welcome! And thank you for reporting the bug!

commit da024a15 (2018-03-20): Fix add_subdirectory() regression

Fixes #46