fkie/catkin_lint #50: Introduce a way to ignore specific errors

koenlek, 2018-06-13:
We want to use `catkin lint` as a hard requirement for CI to pass. As explained [here](http://fkie.github.io/catkin_lint/#limitations), catkin lint has its limitations and output won't be always correct.

It would be good to be able to add a way to define exceptions for certain specific errors, like other linters support.

For example by adding a `.catkinlint_ignore` file (recursively detected, like `.gitignore`) that defines exceptions for a specific package. Or by adding comments like in the CMakeLists.txt or package.xml `

```
add_library(ANPPmsg26         STATIC src/ANPPmsg26.cpp) # CATKIN_LINT_DISABLE_INSTALL_CHECK
```

roehling, 2018-06-13:
This feature has been implemented for quite some time. See:
http://fkie.github.io/catkin_lint/messages/

koenlek, 2018-06-13:
Thanks a lot! I overlooked that. I tried searching for it, but not hard enough apparently.

closed 2018-06-13 by koenlek