fkie/catkin_lint #51: Failing test

sanjayankur31, 2018-06-25:
Hi!

I was just updating the Fedora package to the latest release, and [one of the tests seem to be failing](https://github.com/fkie/catkin_lint/blob/master/test/test_all_checks.py#L204):

```
+ /usr/bin/python3 setup.py test
running test
running egg_info
writing src/catkin_lint.egg-info/PKG-INFO
writing dependency_links to src/catkin_lint.egg-info/dependency_links.txt
writing entry points to src/catkin_lint.egg-info/entry_points.txt
writing requirements to src/catkin_lint.egg-info/requires.txt
writing top-level names to src/catkin_lint.egg-info/top_level.txt
reading manifest file 'src/catkin_lint.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'src/catkin_lint.egg-info/SOURCES.txt'
running build_ext
src/catkin_lint/linter.py:292: DeprecationWarning: invalid escape sequence \$
  if len(arg_tokens) == 1 and re.match("\${[a-z_0-9]+}$", arg_tokens[0][1]):
test_project (test.test_all_checks.AllChecksTest)
Test minimal catkin project for compliance ... ok
runTest (test.test_all_checks.CatkinInvokationTest)
Test catkin_lint invocation on a ROS workspace ... FAIL
test_depends (test.test_checks_build.ChecksBuildTest)
Test dependency checks (POSIX/NT) ... ok
test_dynamic_reconfigure (test.test_checks_build.ChecksBuildTest)
Test checks for dynamic reconfigure scripts (POSIX/NT) ... ok
test_exports (test.test_checks_build.ChecksBuildTest)
Test checks for exported libraries (POSIX/NT) ... ok
test_format2_message_exports (test.test_checks_build.ChecksBuildTest)
Test checks for package format version 2 features ... ok
test_includes (test.test_checks_build.ChecksBuildTest)
Test include_directories() (POSIX/NT) ... ok
test_installs (test.test_checks_build.ChecksBuildTest)
Test installation checks (POSIX/NT) ... ok
test_link_directories (test.test_checks_build.ChecksBuildTest)
Test link_directories() (POSIX/NT) ... ok
test_message_generation (test.test_checks_build.ChecksBuildTest)
Test ROS message generation checks ... ok
test_name_check (test.test_checks_build.ChecksBuildTest)
Test checks for invalid names (POSIX/NT) ... ok
test_plugins (test.test_checks_build.ChecksBuildTest)
Test checks for exported plugins (POSIX/NT) ... ok
test_scripts (test.test_checks_build.ChecksBuildTest)
Test checks for executable scripts (POSIX/NT) ... ok
test_source_files (test.test_checks_build.ChecksBuildTest)
Test add_executable() and add_library() (POSIX/NT) ... ok
test_targets (test.test_checks_build.ChecksBuildTest)
Test checks catkin packages with declared targets (POSIX/NT) ... ok
test_tests (test.test_checks_build.ChecksBuildTest)
Test unit test checks ... ok
test_targets (test.test_checks_cuda.ChecksCudaTest)
Test CUDA checks (POSIX/NT) ... ok
test_fkie_find_package (test.test_checks_fkie.ChecksFkieTest)
Test FKIE specific checks ... ok
test_catkin_build (test.test_checks_manifest.ChecksManifestTest)
Test catkin build system checks ... ok
test_depends (test.test_checks_manifest.ChecksManifestTest)
Test dependency checks for package.xml ... ok
test_export_targets (test.test_checks_manifest.ChecksManifestTest)
Test check for valid exported targets ... ok
test_package_description (test.test_checks_manifest.ChecksManifestTest)
Test check for descriptive package descriptions ... ok
test_cmake_includes (test.test_checks_misc.ChecksMiscTest)
Test CMake includes (POSIX/NT) ... ok
test_cmake_modules (test.test_checks_misc.ChecksMiscTest)
Test check for cmake_modules usage ... ok
test_deprecated (test.test_checks_misc.ChecksMiscTest)
Test check for deprecated features ... ok
test_endblock (test.test_checks_misc.ChecksMiscTest)
Test proper style for CMake code blocks ... ok
test_global_vars (test.test_checks_misc.ChecksMiscTest)
Test global variable checks ... ok
test_minimum_version (test.test_checks_misc.ChecksMiscTest)
Test check for CMake minimum compatible version ... ok
test_project (test.test_checks_misc.ChecksMiscTest)
Test project() ... ok
test_singleton_command (test.test_checks_misc.ChecksMiscTest)
Test check for singleton commands ... ok
test_special_vars (test.test_checks_misc.ChecksMiscTest)
Test checks for proper handling of special variables ... ok
test_setup_with_setup_py (test.test_checks_python.ChecksPythonTest)
Test proper placement and handling of catkin_python_setup() (POSIX/NT) ... ok
test_setup_without_setup_py (test.test_checks_python.ChecksPythonTest)
Test catkin_python_setup() call without setup.py ... ok
test_argparse (test.test_cmake_parser.CMakeParserTest)
Test CMake parser argparse utility function ... ok
test_arguments (test.test_cmake_parser.CMakeParserTest)
Test CMake parser argument parsing ... ok
test_command (test.test_cmake_parser.CMakeParserTest)
Test CMake parser command parsing ... ok
test_comments (test.test_cmake_parser.CMakeParserTest)
Test CMake parser comment handling ... ok
test_empty (test.test_cmake_parser.CMakeParserTest)
Test CMake parser with empty file ... ok
test_foreach (test.test_cmake_parser.CMakeParserTest)
Test CMake parser foreach() loop handling ... ok
test_function (test.test_cmake_parser.CMakeParserTest)
Test CMake parser function definitions ... ok
test_line_columns (test.test_cmake_parser.CMakeParserTest)
Test CMake parser column numbering ... ok
test_line_numbering (test.test_cmake_parser.CMakeParserTest)
Test CMake parser line numbering ... ok
test_macro (test.test_cmake_parser.CMakeParserTest)
Test CMake parser macro expansion ... ok
test_pragma (test.test_cmake_parser.CMakeParserTest)
Test CMake parser catkin_lint pragmas ... ok
test_string (test.test_cmake_parser.CMakeParserTest)
Test CMake parser string parsing ... ok
test_substitution (test.test_cmake_parser.CMakeParserTest)
Test CMake parser variable substitution semantics ... ok
test_foreach (test.test_indentation.IndentationTest)
Test indentation checks for foreach()/endforeach) blocks ... ok
test_if (test.test_indentation.IndentationTest)
Test indentation check for if()/else()/endif() blocks ... ok
test_macro (test.test_indentation.IndentationTest)
Test indentation check for sequences with macro calls ... ok
test_regular (test.test_indentation.IndentationTest)
Test indentation check for regular command sequences ... ok
test_argparse_error (test.test_linter.LinterTest)
Test invalid CMake command arguments ... ok
test_blacklist (test.test_linter.LinterTest)
Test CMake inclusion blacklist (POSIX/NT) ... ok
test_circular_depend (test.test_linter.LinterTest)
Test circular dependencies in custom linter modules ... ok
test_env_var (test.test_linter.LinterTest)
Test environment variable handling ... ok
test_environment (test.test_linter.LinterTest)
Test catkin environment (POSIX/NT) ... ok
test_if (test.test_linter.LinterTest)
Test if()/else()/endif() block handling ... ok
test_include (test.test_linter.LinterTest)
Test edge cases for CMake include ... ok
test_list (test.test_linter.LinterTest)
Test CMake list handling ... ok
test_lower_case (test.test_linter.LinterTest)
Test check for lower-case command names ... ok
test_pragma (test.test_linter.LinterTest)
Test #catkin_lint: pragma handling ... ok
test_subdir (test.test_linter.LinterTest)
Test add_subdirectory() (POSIX/NT) ... ok
test_explained_text (test.test_output.OutputTest)
Test output format for catkin_lint text output with explanations ... ok
test_text (test.test_output.OutputTest)
Test output format for catkin_lint text output ... ok
test_xml (test.test_output.OutputTest)
Test output format for catkin_lint XML output ... ok
test_is_sorted (test.test_util.UtilTest)
Test is_sorted() utility function ... ok
test_word_split (test.test_util.UtilTest)
Test word_split() utility function ... ok
======================================================================
FAIL: runTest (test.test_all_checks.CatkinInvokationTest)
Test catkin_lint invocation on a ROS workspace
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/builddir/build/BUILD/catkin_lint-1.4.21/test/test_all_checks.py", line 205, in runTest
    self.assertEqual(exitcode, 1)
AssertionError: 0 != 1
----------------------------------------------------------------------
Ran 66 tests in 0.520s
FAILED (failures=1)
error: Bad exit status from /var/tmp/rpm-tmp.2JSj0V (%check)
```

Would you know what's causing this by any chance (before I go digging myself :)). I've also attached the complete build log here:
[build-log.txt](https://github.com/fkie/catkin_lint/files/2132552/build-log.txt)

Thanks!

roehling, 2018-06-25:
Turns out that when the `rosdep2` module is not installed, `catkin_lint` will ignore missing dependencies (and print a message to that effect), but a few tests rely on `catkin_lint` failing for a missing dependency.

I have fixed that problem with commit ad6ff3047fea6c696144ab27729614eb75d3fc43

sanjayankur31, 2018-06-25:
Thanks. I'll use the patch for the time being. Is rosdep2 different from rosdep (I can only find this on pypi: https://pypi.org/project/rosdep/)? (If it is, we'll have to package it for Fedora first)

roehling, 2018-06-25:
No, it's the same, but for some reason, the module is called rosdep2 while the package is just called rosdep.