fkie/catkin_lint #58: add a check if pkg folder name matches pkg name

koenlek, 2018-10-03:
We would like to enforce that a package `foo` is stored in a folder called `foo`.  The current `project_name` check only checks if package.xml and CMakeLists.txt match, but not if the folder name matches. The description in the documentation suggests that it does that too, but apparently it doesn't as I just tested it with python-catkin-lint debian 1.5.3-1~16.04.

From http://fkie.github.io/catkin_lint/messages/#project-name-name-differs-from-package-name :

> project name 'name' differs from package name
>
>    ID: project_name
>    Severity: error
>    Explanation: The CMake project name must be identical to the package name. For backwards compatibility reasons, both names should also be identical to the name of the source folder that contains the package.

Checking the source folder name too would be great. I'd like to see it as an error, but I can imagine that might upset some people, so warning would be ok too. But if it's a warning, I'll have a practical issue. How can I make our CI check for all errors (except those that are ignored via `#catkin_lint: ignore ` or `--ignore`) + that one specific warning that checks if folder name matches pkg name and treat only that warning as strict?

gavanderhoorn, 2018-10-03:
Project name, folder name and package name are three different things.

project should correspond to package, but folder name can be anything. I don't remember there being a requirement for the latter to be identical to the package name, but it is of course a good convention.

koenlek, 2018-10-03:
True, they are 3 different things. I prefer to keep folder name the same as package name by convention, as a mismatch can lead to confusion within a team, is my experience. Hence, I prefer to be able to enforce this convention using a linter, helping us to keep our codebase clean and organized.

gavanderhoorn, 2018-10-03:
:+1: to the general idea. I just commented as you seemed to imply that keeping all three in sync is a rule or a requirement somewhere.

roehling, 2018-10-05:
Back in the old rosmake days before catkin, the folder name was required to be the same as the package name. It's probably a good idea to check for this. :+1: 

I also like the idea of customizing the severity category of diagnostic messages (like turning warnings into errors), but I'll need to think about this for a while and see if I can implement it without unintended side effects. Some messages already change severity on certain conditions (e.g. uninstalled nodes will be an error or a warning, depending on whether there are other install() commands or not).

koenlek, 2018-11-06:
@roehling : are you still considering implementing a warning like `directory_name` for checking if directory name and pkg name (in package.xml) are matching? 

Regarding 'customizing the severity category of diagnostic message': how about this? 
Add a `--only` argument that allows you to *only* check for a certain ID or list of IDs. Then, you can make it check strictly for those IDs. Then our CI lint job would exist of two commands:

```
catkin lint -W0 .
catkin lint --only directory_name some_other_warning_thats_very_important_to_us --strict -W1 .
```

You could add it like this (I like the `nargs='+'`, a list or arg values can always be terminated by the next arg or by closing it with a `--` at the end):

```
parser.add_argument('--only',    help='only check for this ID or list of IDs', type=str, nargs='+')
```

Would be really helpful :)

roehling, 2018-11-19:
I added three new command line options (--error, --warning, --notice) which work similar to --ignore, i.e. you can force the severity of a report as you see fit.

Ex: `catkin_lint . --error package_path_name`

closed 2018-11-20 by roehling

koenlek, 2018-11-21:
That is great! Thanks a lot. Will give it a try as soon as it is released via the apt repos

commit c04d2ae2 (2018-11-20): 1.5.4

Fixes #58
Fixes #60