fkie/catkin_lint #61: allow variable names in cmake version checks

rhaschke, 2018-12-07:
catkin_lint suggests to change
`if(rviz_QT_VERSION VERSION_LESS "5")`
into
`if("${rviz_QT_VERSION}" VERSION_LESS "5")`

with the following `notice: operands for operator VERSION_LESS should be quoted strings`.

However, using a variable name directly, without quoting and resolving is perfectly valid and IMHO much more readable. Could you improve this check, please?

roehling, 2019-01-11:
I improved the check, because the main goal was to prevent constructs like

    set(var1 OFF)
    set(var2 var1)
    # ...
    if(${var2})
        # ...
    endif()

In this case, `${var2}` resolves to `var1`, and `var1` is then treated as variable and resolved *again*.

closed 2019-01-11 by roehling

rhaschke, 2019-01-14:
Isn't that something we would like to happen?

roehling, 2019-01-14:
If you are very adept with CMake, maybe. Most people I know are surprised by this behavior, which is exactly why `catkin_lint` warns about it in the first place.

commit 059fcdea (2019-01-11): Make the unquoted string operation notice more precise

Only trigger the notice if an unquoted variable resolution ${...} is
used, which may have unintended results.

Fixes #61