fkie/catkin_lint #62: spurious errors?

rhaschke, 2019-02-07:
Running `catkin_lint` on [`moveit_core`](https://github.com/ros-planning/moveit/tree/melodic-devel/moveit_core), I get the following error messages:
```
moveit_core: CMakeLists.txt(89): error: catkin_package() exports pkg-config module 'LIBFCL'
moveit_core: CMakeLists.txt(154): error: include_directories() needs missing directory '/include'
moveit_core: collision_detection_fcl/CMakeLists.txt(20): error: duplicate find_package(moveit_resources)
moveit_core: collision_distance_field/CMakeLists.txt(23): error: duplicate find_package(moveit_resources)
moveit_core: constraint_samplers/CMakeLists.txt(30): error: duplicate find_package(moveit_resources)
moveit_core: kinematic_constraints/CMakeLists.txt(20): error: duplicate find_package(moveit_resources)
moveit_core: planning_scene/CMakeLists.txt(25): error: duplicate find_package(moveit_resources)
moveit_core: robot_state/CMakeLists.txt(26): error: duplicate find_package(moveit_resources)
moveit_core: trajectory_processing/CMakeLists.txt(20): error: duplicate find_package(moveit_resources)
moveit_core: utils/CMakeLists.txt(20): error: duplicate find_package(moveit_resources)
```
I'm not sure why a duplicate `find_package` should be an error at all - it doesn't harm, does it?
In this particular case, however, it's not even duplicated! All calls to `find_package(moveit_resources)` are in sub-directories of the main package only and thus don't "see" each other.

Also, why shouldn't it be possible to export a pkg-config package in `catkin_package()`? catkin just collects the content of `NAME_INCLUDE_DIRS` and `NAME_LIBRARIES` to setup the package's cmake-config file. These variables are well-defined for pkg-config packages as well.

Finally, the variable value of [`CATKIN_DEVEL_PREFIX` seems not to be used](https://github.com/ros-planning/moveit/blob/8121e4690190223ef18b808112ca907d50761563/moveit_core/CMakeLists.txt#L59), leading to the include error:
`moveit_core: CMakeLists.txt(154): error: include_directories() needs missing directory '/include'`

rhaschke, 2019-02-07:
The [explanation for duplicate find_package](http://fkie.github.io/catkin_lint/messages/#duplicate-find_packagepkg) is wrong: A second call to find_package will be fully processed, considering different arguments. However, it's true that subsequent calls to `find_library` or `find_path` will be cached and not executed twice.
For example, it's perfectly valid to call `find_package(catkin)` twice:
```
find_package(catkin REQUIRED)
message("1: ${catkin_INCLUDE_DIRS}")
find_package(catkin REQUIRED COMPONENTS roscpp)
message("2: ${catkin_INCLUDE_DIRS}")
```
While, after the first call `catkin_INCLUDE_DIRS` will be empty, after the second call it will list the new components' include paths.

I think, `catkin_lint` should compare arguments, particularly listed `COMPONENTS`, and only complain if the component list has overlaps to previous calls. Also, the error should be downgraded to a warning.

roehling, 2019-02-07:
I'm currently on sick leave, so I can't really test anything, but it looks like there is a bug in `catkin_lint` regarding the subdirectory scope. Also, I'm not sure what's going on with the CATKIN_DEVEL_PREFIX variable, so I'm going to look into this as well.

Everything else you brought up is pretty much covered by the extended explanations of `catkin_lint` (which you can see with the `--explain` option), but let me elaborate a bit.

Duplicate `find_package` calls will be ignored by CMake, regardless of the options given, since the first successful call will be cached and reused. So if you do something like
```
find_package(catkin REQUIRED)
find_package(catkin REQUIRED COMPONENTS some_catkin_package)
```
Your package will mysteriously fail to link against `some_catkin_package`, without any hints from CMake. Given that there are very few advanced usages of multiple finds which usually involve NO_DEFAULT_* options to change the search order, I find the error justified (and like all diagnostics, it's overrideable if your preference differs).

Regarding `pkg-config`, it works very differently from CMake, since it will merely give you linker flags and not the actual libraries with paths. For a real-life example, I currently have both OpenCV 3 and OpenCV 4 installed on my system, and I can conveniently select between those with `find_package(OpenCV 3)` and `find_package(OpenCV 4)`. In contrast, `pkg-config opencv --libs` will give me a bunch of -lopencv_* flags that will always use whichever version happens to come first in my LD_LIBRARY_PATH, destroying my ability to choose. I recommend [this](https://gitlab.kitware.com/cmake/community/wikis/doc/tutorials/How-To-Find-Libraries#piggybacking-on-pkg-config) for further reading if the library has no dedicated CMake config script

roehling, 2019-02-07:
It's possible that starting with CMake 3, the semantics of `find_package()` have changed, or the catkin implementation has improved over the years. I know for a fact that your code used to fail, I tried it myself a couple years ago.

rhaschke, 2019-02-07:
> It's possible that starting with CMake 3, the semantics of find_package() have changed.

Maybe, but I can't believe so.

> or the catkin implementation has improved over the years.

Couldn't find any hint for this as well. Looks like COMPONENTS were processed unconditionally for ages.

> I know for a fact that your code used to fail, I tried it myself a couple years ago.

Probably it's time to change this.
Thanks for your quick answer and get well!

roehling, 2019-02-12:
It looks to me that you did not check with the latest release, since I cannot reproduce the CATKIN_DEVEL_PREFIX issue with version 1.5.5.

rhaschke, 2019-02-12:
> It looks to me that you did not check with the latest release, since I cannot reproduce the CATKIN_DEVEL_PREFIX issue with version 1.5.5.

Indeed, I'm using version 1.5.2, which is the latest release provided from the `ros-shadow-fixed` repository. Obviously, you didn't released via `bloom` recently. Please can you do so as well?
I think this is the most common channel, people are accessing this package with.

rhaschke, 2019-02-12:
I recently fixed most of the catkin_lint reported issues in MoveIt (https://github.com/ros-planning/moveit/pull/1341). All the spurious errors, I reported here, are explicitly suppressed to pass the automatic Travis test.

roehling, 2019-02-12:
AFAIK, the `catkin_lint` package gets synced manually from my PPA once in a while. I'll open a ticket for an update after the 1.5.6 release, which I'll do as soon as this issue is resolved.

roehling, 2019-02-12:
That's why I just added a `--show-suppressed` flag, which makes `catkin_lint` report them anyway ;-)
So these are the errors you would get without any ignores.

roehling, 2019-02-12:
Just to wrap this up, if you have no objections, I'll consider the spurious errors to be sufficiently dealt with now.

rhaschke, 2019-02-12:
> That's why I just added a --show-suppressed flag, which makes catkin_lint report them anyway ;-)

I see. I fully agree with you, that the remaining warnings are legit. Looks like your master branch resolved this issue. Thanks a lot.

closed 2019-02-12 by roehling

roehling, 2019-02-13:
FYI ros/rosdistro#20283