fkie/catkin_lint #65: Incorrectly reports `uninstalled_plugin` if plugin XML file is not in root of package

jarvisschultz, 2019-03-04:
Using relative paths for plugin XML files seems to well in a `package.xml` file and in a `CMakeLists.txt` file; I've found numerous example packages where this is done. For example, in a `package.xml` you might have

```xml
<export>
  <gazebo_ros_control plugin="${prefix}/config/robot/robot_hw_sim_plugins.xml"/>
</export>
```
and then in the `CMakeLists.txt` you could have

```cmake
install(
  FILES config/robot/robot_hw_sim_plugins.xml
  DESTINATION ${CATKIN_PACKAGE_SHARE_DESTINATION}
  )
```
Unfortunately, this throws an `uninstalled_plugin` error. I believe this is due to a call to `posixpath.basename` when [building a list of the installed files](https://github.com/fkie/catkin_lint/blob/90594e3005152c9c924b8fd8a291e22bc213f559/src/catkin_lint/checks/build.py#L474) which strips all leading paths and leaves you with just the filename and extension.

I removed the call to `posixpath.basename`, and that fixed my error. That said, I'm unsure if there would be other unforeseen consequences related to this change.

roehling, 2019-03-04:
The `install()` command will copy the source file directly to the specified destination, so your `robot_hw_sim_plugins.xml` will end up in `share/PACKAGE`, and not in `share/PACKAGE/config/robot` where it is expected according to the plugin declaration of your `package.xml`.

roehling, 2019-03-04:
The command
```cmake
install(
  FILES config/robot/robot_hw_sim_plugins.xml
  DESTINATION ${CATKIN_PACKAGE_SHARE_DESTINATION}/config/robot
  )
```
should work as expected and not be flagged by `catkin_lint`.

jarvisschultz, 2019-03-05:
You are totally correct. Ugh! Sorry for the distraction, and thanks for the tip.

closed 2019-03-05 by jarvisschultz