fkie/catkin_lint #66: Error for Eigen3 export

aPonza, 2019-03-29:
```
$ catkin_lint -W2 --explain --pkg <my_pkg>
<my_pkg>: CMakeLists.txt(105): error: catkin_package() exports non-package include path
```

<my_pkg>/CMakeLists.txt
```
catkin_package(
  INCLUDE_DIRS include ${EIGEN3_INCLUDE_DIRS}
  LIBRARIES ${PROJECT_NAME}
  CATKIN_DEPENDS
    shape_msgs
#  DEPENDS
)
```

However, according to [this page](http://wiki.ros.org/jade/Migration#Eigen_CMake_Module_in_cmake_modules) this is the correct way to proceed after having done a `find_package(Eigen3 REQUIRED)`. This is Jade documentation, I'm on Kinetic so it all might be wrong, but it's also the most updated source I have found (from [here](https://answers.ros.org/question/253588/ros-kinetic-eigen/)).

gavanderhoorn, 2019-03-29:
Shouldn't that be:

```cmake
catkin_package(
  INCLUDE_DIRS include
  LIBRARIES ${PROJECT_NAME}
  CATKIN_DEPENDS
    shape_msgs
  DEPENDS EIGEN3
)
```

aPonza, 2019-03-29:
Hah, thanks @gavanderhoorn, I had tried with `Eigen3` which was giving me a warning, and had then found the page I linked, but you are right. Cheers!

closed 2019-03-29 by aPonza