fkie/catkin_lint #67: Incorrect ID for target 'target' is not installed

ThomasDelbaere, 2019-06-05:
After getting a warning about a target not installed, I referred to the documentation [here](http://fkie.github.io/catkin_lint/messages/#target-target-is-not-installed) to get the ID of the message and ignore it with : 
`#catkin_lint: ignore uninstalled_target`

The warning was still there but thanks to the argument --explain I got this output : 
`     * You can ignore this problem with --ignore missing_install_target`
And now the warning is properly ignored so the documentation should be updated with `missing_install_target` instead of `uninstalled_target`.

Note : I dug into the code and found in `output.py` this : 
`fd.write("     * You can ignore this problem with --ignore %s\n" % msg.id.lower())`
But when I checked in `diagnostics.py` where all the message ids are defined there was no `MISSIN_INSTALL_TARGET` as expected but only `UNINSTALLED_TARGET` so I couldn't figure out how the argument --explain could possibly return the correct output.

roehling, 2019-06-05:
For the latest 1.6 release, I merged a few very similar IDs and got rid of a few deprecated ones, so my guess is that you are using an older version. You can check with `catkin_lint --version`.

ThomasDelbaere, 2019-06-05:
You're absolutly right, I thought after running an apt-get update and upgrade I would have the latest version but apparently not.
Sorry for the mistake, now everything is fine.

closed 2019-06-05 by ThomasDelbaere