fkie/catkin_lint #73: spurious error for eigenpy not being a catkin package

rhaschke, 2019-11-15:
With its recent release (https://github.com/ros/rosdistro/pull/22935), `eigenpy` changed its `package.xml` to include:
```xml
<export>
    <build_type>cmake</build_type>
</export>
```
Because of this declaration of eigenpy as building with pure cmake, catkin_lint now complains:
`moveit_ros_planning_interface: CMakeLists.txt(10): error: 'eigenpy' in find_package(catkin) is not a catkin package`

However, it's still actually a catkin package, isn't it?
- it installs a package.xml
- the cmake file exports `_INCLUDE_DIRS` and `_LIBRARIES` variables.

gavanderhoorn, 2019-11-15:
> However, it's still actually a catkin package, isn't it?

Perhaps not: according to [REP 127 - Specification of package manifest format: build_type](https://www.ros.org/reps/rep-0127.html#build-type):

>  Typical value is `cmake` to denote a plain CMake package, but any string is allowed.

rhaschke, 2019-11-15:
As I said, the package builds as a plain cmake package, but when installed, it's found as a normal catkin package and can be used as such.

gavanderhoorn, 2019-11-15:
Just quoting the REP here.

Packages with `export/build_type` set to `cmake` are not considered Catkin packages in my experience. That is exactly what the point of the `<build_type>cmake</build_type>` is.

rhaschke, 2019-11-15:
I don't agree. Why should I have a package.xml if the package isn't recognized by catkin?
Actually, it _is_ recognized by catkin: 
- `roscd eigenpy` brings you to the package folder
- the package is listed by `catkin_topological_order`
- `find_package(catkin COMPONENTS eigenpy)` finds the package
- etc.

gavanderhoorn, 2019-11-15:
All of what you write is because there is a `package.xml`, which is what `rospkg`/`catkin_pkg` needs to consider *the directory* a ROS package.

`catkin_lint` looks at the build script.

Installing a `package.xml` for a non-Catkin/3rd party package is a simple trick to make something buildable in a Catkin workspace/workflow. It doesn't automagically make something a Catkin package.

See also [REP 136](https://www.ros.org/reps/rep-0136.html).

But tbh, I don't really see why we need to debate this. Let's wait on @roehling.

roehling, 2019-11-16:
@gavanderhoorn is right.

Amusingly, this is a direct consequence of issue #71, where the opposite problem occured in the moveit community, and the reason why @wxmerkt added the new `build_type` to the `eigenpy` package.

You need to distinguish between the package toolchain and the build system.  In the old days, ROS had a build system called rosbuild and kept the package metadata in a `manifest.xml` file. With the introduction of catkin, the package format was substantially overhauled (and the manifest file renamed to `package.xml`). This simultaneous change is probably where most of the confusion about the package format versus the build system originates.

The catkin build system piggybacks on CMake, but requires packages to do their build in a very specific way in order to support various catkin features (such as the `devel` tree, ROS message definitions, and so on). This is the primary reason why you need to call `find_package(catkin)` and `catkin_package()`. It is also the reason why `catkin_lint` can be a bit of a hard-ass when you deviate from the canonical way a  `CMakeList.txt` is supposed to be organized.

However, the package toolchain allows for alternative build systems, for instance "plain" CMake builds, which essentially turns a package into a system dependency that is built from source. A very popular example for this used to be the OpenCV 3 package that was part of the ROS distribution before it became part of the official Ubuntu release. Another example is the [WebRTC package](https://github.com/RobotWebTools/webrtc_ros/tree/develop/webrtc), which is actually not even built with CMake (it still provides a CMake configuration though, so you can use `find_package()` on it).

To summarize: everything with a `package.xml` is a ROS package. If and only if it uses the catkin build system, it's a catkin package. For ROS 1, this is the default setting and can be overriden with the `build_type` tag in the manifest.

wxmerkt, 2019-11-16:
Thank you for the detailed explanation @gavanderhoorn @roehling.

@rhaschke I am aware that this REP 136 released third-party package is _not_ a catkin package. We added the `eigenpyConfig.cmake` to avoid a breakage in MoveIt (as a work-around) and avoid the MoveIt release. Catkin duly finds the package - but technically this is not a catkin package. 

The correct usage for finding is would be:

```cmake
find_package(eigenpy REQUIRED)
```
or
```cmake
find_package(PkgConfig REQUIRED)
pkg_check_modules(eigenpy REQUIRED eigenpy)
```

So I don't think there is anything wrong with the linter. is it possible to silence the warning for now?

roehling, 2019-11-16:
You can silence the warning by preceding the `find_package(catkin)` line with a comment line that reads `#catkin_lint: ignore_once no_catkin_component`. For more details and alternative workarounds, take a look at the [relevant section of the catkin_lint documentation](https://fkie.github.io/catkin_lint/messages/).

closed 2019-11-17 by rhaschke