fkie/catkin_lint #77: Rule 'plugins'?

gavanderhoorn, 2020-01-16:
Something which came up in a discussion here: would it be possible (or perhaps better: interesting) to add a sort of plugin system to `catkin_lint` which would allow packages to contribute additional and specialised rules which `catkin_lint` would then add to its own rule set and check for?

Certain labs and companies have specific in-house styles and conventions for packaging (ROS) components, and it would be great if `catkin_lint` could check and flag violations against such conventions.

I haven't checked the feasibility, but would that be something that would be possible with the current architecture of `catkin_lint`?

roehling, 2020-01-17:
Actually, this has been possible since day one, even though I'm not aware of anyone ever using this feature: http://fkie.github.io/catkin_lint/api/

gavanderhoorn, 2020-01-17:
Nice.

Not sure how I've missed that all these years.

Thanks.

closed 2020-01-17 by gavanderhoorn

gavanderhoorn, 2020-01-17:
Quick question: IIUC the plugin linters "just" need to be present on the `PYTHONPATH`, correct? Is that sufficient for them to be discoverable and loadable by `catkin_lint -c ..`?

roehling, 2020-01-17:
Everything in the checks module counts as example ;-)

One thing that is missing is that you cannot yet add custom messages. I'll have to go ahead and add something to the API for that.

Oh, and the `-c` option replaces the default modules. But you can have a comma-separated list with multiple modules, so something like `-c all,mymodule.myfunc` should work.

roehling, 2020-01-23:
I went ahead and added something to the API for that.

gavanderhoorn, 2020-01-23:
That's really nice. Thanks for this.

:+1: