fkie/catkin_lint #78: yaml-cpp ambivalence: YAML_CPP or yaml-cpp?

gavanderhoorn, 2020-03-05:
Context: https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/pull/132.

Related context: https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/pull/42.

The issue is that it would appear to be impossible to appease both `catkin_lint` *and* `catkin_package(..)` when it comes to `yaml-cpp` :)

The variables defined on the CMake side are prefixed with `YAML_CPP`. The CMake configuration files (in `/usr/lib/x86_64-linux-gnu/cmake/yaml-cpp` on my system) however are prefixed with `yaml-cpp` (lowercase).

Listing `YAML_CPP` in the `DEPENDS` list makes `catkin_package(..)` happy (as it can now find the required variables). But `catkin_lint` doesn't like it, as -- understandably -- it's looking for a `find_package(YAML_CPP ..)` (which doesn't exist). Other way around (listing `yaml-cpp`) makes `catkin_lint` happy, but now `catkin_package(..)` obviously can't find the needed variables.

Tbh this is a problem on yaml-cpp's side, but I cannot change it.

I'm looking for the 'nicest' way to resolve this situation. I've already considered adding alias variables (ie: `yaml-cpp_INCLUDE_DIRS` et al.) but it isn't too nice. An alternative could be to disable `catkin_lint` checks for this particular line.

roehling, 2020-03-05:
I just grepped through my Linux installation for all CMake modules and their variables, and I found yaml-cpp to be the one exception where the variable prefix is neither "Name" nor "NAME". Thus, I went with the simple (albeit ugly) solution, and added a special case.

It's not perfect, but it will do. I'm amazed how far this ham-fisted approach has carried this tool anyway. ;-)

closed 2020-03-05 by roehling

gavanderhoorn, 2020-03-05:
Ha :)

Definitely not perfect, but `yaml-cpp` appears to be an exception.

thanks for the quick resolution. :+1:

commit 615dbd70 (2020-03-05): Add a special case for yaml-cpp

Closes #78