fkie/catkin_lint #80: Include dirs var becomes strange character

mortenfyhn, 2020-03-30:
Hi again!

In one of my projects, I'm getting an error like this
```
<my project>: CMakeLists.txt(60): warning: target_include_directories() has foreign PUBLIC path '${_INCLUDE_DIRS}'
```
where the strange character before `_INCLUDE_DIRS` shows up differently in different places.

In my terminal it shows up like this:

[image]

and if I copy from my terminal into my editor, it looks like this:

[image]

In the CMakeLists.txt file, it's just `${MAVLINK_INCLUDE_DIRS}`.

I think this might be a false positive caused by some parsing error.

roehling, 2020-03-30:
Yeah, that's a bug I introduced with the last release. And of course it is the one thing I did not test properly because I was lazy and thought it would be working.

roehling, 2020-03-30:
For the record, it is not a false positive, it's just a garbled output message. What it means is that you should avoid putting find_packaged() paths into INTERFACE and PUBLIC includes, because if you export the targets, CMake will hard-code those paths in the exported config.

mortenfyhn, 2020-03-30:
I see, thank you! Sounds familiar.

I guess I still have some things to learn about cmake and catkin-cmake, but that's part of the reason why this project is so useful.

commit a97518c3 (2020-03-30): Fix path reporting

Well, this happens if you think you can do it without proper testing. At
least it should work as expected now.

Closes #80
closed 2020-03-30 by roehling