fkie/catkin_lint #82: Ignore things that git ignores

mjsobrep, 2020-06-05:
It would be great if a flag could be passed to tell the system to ignore gitignored files/folders when searching recursively and running the tests.

roehling, 2020-06-05:
`catkin_lint` ignores any subtrees with a `CATKIN_IGNORE` file, just as `catkin` does. Do you have a particular use case in mind where this is not sufficient?

mjsobrep, 2020-06-05:
I have some non-ros web code that lives next to my ros code. Inside of node_modules folders there are files that are executable, but obviously  not installed as part of my ros system. It would be great to ignore all of those files. I though that using the ignore-paths arg would do that, but it looks like it just specifies the places to not look for ros packages. Ideally catkin_lint wouldn't see the files in the gitignore at all. 

It is possible that co-locating code the way that I do is an awful practice or that it is beyond the scope of catkin_lint to do this and I should be going into a clean environment and cleaning out everything in the gitignore before running catkin_lint.

roehling, 2020-06-10:
First of all, going over the code I realized that `CATKIN_IGNORE`will not be honored when crawling for executable files, so I'm going to add that.

Secondly, I would prefer not to add any Git-specific semantics, as `catkin_lint` is not related at all to version control, and it would be a bottomless pit (what about Mercurial? SVN? Bazaar?). 

I suggest you add the following key to your `package.json`:

    "scripts": {
        "postinstall": "touch node_modules/CATKIN_IGNORE || true"
    }

This will automatically add a `CATKIN_IGNORE` file after `npm install`, which will (after the next release) achieve what you need.

commit 296d206f (2020-06-17): Honor CATKIN_IGNORE when enumerating files in a package

Even though CATKIN_IGNORE is primarily intended to ignore a subtree when
crawling for packages, catkin_lint will now ignore subtrees with a
CATKIN_IGNORE file for all operations which attempt to analyse all files
in a package.

Fixes #82