fkie/catkin_lint #84: `if (${CATKIN_ENABLE_TESTING})` not supported

peci1, 2020-06-09:
I decided to use the more bulletproof `if (${CATKIN_ENABLE_TESTING})`, but I keep getting `UNGUARDED_TEST_CMD` error.

roehling, 2020-06-09:
Why exactly do you think this is more bulletproof? It is vulnerable to [spurious variable expansions](https://cmake.org/cmake/help/latest/command/if.html#variable-expansion).

peci1, 2020-06-09:
Hmm, I got the impression that the `${}` approach is the preferred one and the other one is going to be abandoned one time... But apparently, I was wrong.

It just seems like a big inconsistency to me... And there is https://cmake.org/cmake/help/latest/policy/CMP0054.html which aims towards the `${}` version without spurious expansions.

roehling, 2020-06-10:
> And there is https://cmake.org/cmake/help/latest/policy/CMP0054.html which aims towards the `${}` version without spurious expansions.

That is true, but you'd have to use `if("${CATKIN_ENABLE_TESTING}")` then (note the extra quotes). I still find `if(CATKIN_ENABLE_TESTING)` the simplest and most aesthetically pleasing, but I made `catkin_lint` work with both versions.

commit 0b1341f3 (2020-06-10): Make check for CATKIN_ENABLE_TESTING more robust

Fixes #84
closed 2020-06-10 by roehling