fkie/catkin_lint #85: 'launch_depend` and parent depends

comkieffer, 2020-06-25:
Say I have two packages: 

- `robot_description` which has an exec_depend on `xacro`
- `robot_gazebo` with an exec_depend on `robot_description`

If `robot_gazebo` uses `xacro` in a launch file the `launch_depend` check will fail. This seems like a false positive since `xacro` is already required by `robot_description`. 

If this makes sense to you, I might take a stab at fixing it.

roehling, 2020-06-25:
This is not a false positive, because the transitive dependency is not enough. If `robot_description` ever stopped to depend on `xacro` (maybe because someone wrote `xacro2`), the `robot_gazebo` package would break, because it still uses the `xacro` package.

comkieffer, 2020-06-26:
Thanks, I hadn't thought of it in that light.

closed 2020-06-30 by roehling