fkie/catkin_lint #86: Installing via pip2 on Xenial is broken

rhaschke, 2020-06-29:
Installing catkin_lint via pip2 on Xenial is broken
```
# pip2 install catkin_lint 
Collecting catkin_lint
  Downloading https://files.pythonhosted.org/packages/cd/43/71f9ae8bb54b0e2fc2d67dc7a4ab545ac78ab79c66592b71180e0cab344b/catkin_lint-1.6.10-py2.py3-none-any.whl (60kB)
    100% |████████████████████████████████| 61kB 832kB/s 
Collecting configparser; python_version < "3" (from catkin_lint)
  Downloading https://files.pythonhosted.org/packages/e5/7c/d4ccbcde76b4eea8cbd73b67b88c72578e8b4944d1270021596e80b13deb/configparser-5.0.0.tar.gz (75kB)
    100% |████████████████████████████████| 81kB 1.7MB/s 
  Running setup.py (path:/tmp/pip-build-jJ5Mic/configparser/setup.py) egg_info for package configparser produced metadata for project name unknown. Fix your #egg=configparser fragments.
Requirement already satisfied (use --upgrade to upgrade): catkin-pkg in /usr/lib/python2.7/dist-packages (from catkin_lint)
Collecting lxml (from catkin_lint)
  Downloading https://files.pythonhosted.org/packages/7a/ac/e5caaa241de06024766872714c38d14e5f885dc453a4b8f9e6463b67c164/lxml-4.5.1-cp27-cp27mu-manylinux1_x86_64.whl (5.5MB)
    100% |████████████████████████████████| 5.5MB 234kB/s 
Requirement already satisfied (use --upgrade to upgrade): docutils in /usr/lib/python2.7/dist-packages (from catkin-pkg->catkin_lint)
Requirement already satisfied (use --upgrade to upgrade): pyparsing in /usr/lib/python2.7/dist-packages (from catkin-pkg->catkin_lint)
Requirement already satisfied (use --upgrade to upgrade): python-dateutil in /usr/lib/python2.7/dist-packages (from catkin-pkg->catkin_lint)
Building wheels for collected packages: unknown, unknown
  Running setup.py bdist_wheel for unknown ... done
  Stored in directory: /root/.cache/pip/wheels/42/ac/58/279787ca533d48f087c9abb75d7e53396e37e5e32674746e81
  Running setup.py bdist_wheel for unknown ... error
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-jJ5Mic/unknown/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" bdist_wheel -d /tmp/tmpQCQQV1pip-wheel- --python-tag cp27:
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
  IOError: [Errno 2] No such file or directory: '/tmp/pip-build-jJ5Mic/unknown/setup.py'
  
  ----------------------------------------
  Failed building wheel for unknown
  Running setup.py clean for unknown
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-jJ5Mic/unknown/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" clean --all:
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
  IOError: [Errno 2] No such file or directory: '/tmp/pip-build-jJ5Mic/unknown/setup.py'
  
  ----------------------------------------
  Failed cleaning build dir for unknown
Successfully built unknown
Failed to build unknown
Installing collected packages: unknown, lxml, catkin-lint
Successfully installed catkin-lint-1.6.10 lxml-4.5.1 unknown-0.0.0
You are using pip version 8.1.1, however version 20.1.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
```

rhaschke, 2020-06-29:
Installing configparser 4.0.2 instead of 5.0.0 seems to fix the issue.

roehling, 2020-06-30:
I could not replicate this in a pristine virtual environment (see below), so I suspect this is related to the rather ancient `pip` version that is shipped with Ubuntu 16.04.

    (1)roehling@roehling:~/venv$ virtualenv --version
    virtualenv 20.0.23 from /home/roehling/.local/lib/python2.7/site-packages/virtualenv/__init__.pyc

    (2)roehling@roehling:~/venv$ virtualenv catkin_lint_test
    created virtual environment CPython2.7.17.final.0-64 in 126ms
      creator CPython2Posix(dest=/home/roehling/venv/catkin_lint_test, clear=False, global=False)
      seeder FromAppData(download=False, pip=latest, setuptools=latest, wheel=latest, via=copy, app_data_dir=/home/roehling/.local/share/virtualenv/seed-app-data/v1.0.1)
      activators PythonActivator,CShellActivator,FishActivator,PowerShellActivator,BashActivator

    (3)roehling@roehling:~/venv$ source catkin_lint_test/bin/activate

    (catkin_lint_test) (4)roehling@roehling:~/venv$ python --version
    Python 2.7.17

    (catkin_lint_test) (5)roehling@roehling:~/venv$ pip --version
    pip 20.1.1 from /home/roehling/venv/catkin_lint_test/lib/python2.7/site-packages/pip (python 2.7)

    (catkin_lint_test) (6)roehling@roehling:~/venv$ pip list --user
    DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support

    (catkin_lint_test) (7)roehling@roehling:~/venv$ pip install catkin-lint
    DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
    Collecting catkin-lint
      Downloading catkin_lint-1.6.10-py2.py3-none-any.whl (60 kB)
         |████████████████████████████████| 60 kB 529 kB/s 
    Collecting configparser; python_version < "3"
      Downloading configparser-4.0.2-py2.py3-none-any.whl (22 kB)
    Collecting catkin-pkg
      Downloading catkin_pkg-0.4.22.tar.gz (62 kB)
         |████████████████████████████████| 62 kB 1.0 MB/s 
    Collecting lxml
      Downloading lxml-4.5.1-cp27-cp27mu-manylinux1_x86_64.whl (5.5 MB)
         |████████████████████████████████| 5.5 MB 9.4 MB/s 
    Collecting docutils
      Downloading docutils-0.16-py2.py3-none-any.whl (548 kB)
         |████████████████████████████████| 548 kB 5.7 MB/s 
    Collecting python-dateutil
      Downloading python_dateutil-2.8.1-py2.py3-none-any.whl (227 kB)
         |████████████████████████████████| 227 kB 5.1 MB/s 
    Collecting pyparsing
      Downloading pyparsing-2.4.7-py2.py3-none-any.whl (67 kB)
         |████████████████████████████████| 67 kB 4.7 MB/s 
    Collecting six>=1.5
      Downloading six-1.15.0-py2.py3-none-any.whl (10 kB)
    Building wheels for collected packages: catkin-pkg
      Building wheel for catkin-pkg (setup.py) ... done
      Created wheel for catkin-pkg: filename=catkin_pkg-0.4.22-py2-none-any.whl size=73733 sha256=1400837c110d3304d71d500a3a098e41e8ab93819107ae80d9e2e3bf76a2f624
      Stored in directory: /home/roehling/.cache/pip/wheels/c0/d4/35/6c3d858f3b9e76ec757e915905e1ad74e054befe488cb3d9ba
    Successfully built catkin-pkg
    Installing collected packages: configparser, docutils, six, python-dateutil, pyparsing, catkin-pkg, lxml, catkin-lint
    Successfully installed catkin-lint-1.6.10 catkin-pkg-0.4.22 configparser-4.0.2 docutils-0.16 lxml-4.5.1 pyparsing-2.4.7 python-dateutil-2.8.1 six-1.15.0

    (catkin_lint_test) (8)roehling@roehling:~/venv$ which catkin_lint
    /home/roehling/venv/catkin_lint_test/bin/catkin_lint

    (catkin_lint_test) (9)roehling@roehling:~/venv$ catkin_lint --version
    1.6.10

roehling, 2020-06-30:
> Installing configparser 4.0.2 instead of 5.0.0 seems to fix the issue.

configparser-5.x is Python 3 only, and apparently pip 8.x does not realize that. I just uploaded a post release to PyPI that requires `configparser < 5`. I hope that fixes the issue.

rhaschke, 2020-06-30:
Thanks, this solves the issue on 16.04. Indeed, I was using the default pip 8.x of Ubuntu 16.04.

closed 2020-06-30 by rhaschke