fkie/catkin_lint #87: False positive of unused_include_path

wmmc88, 2020-08-30:
In my project, I'm getting the following `catkin_lint` error:
`include path 'path' is exported but not used for the build`

According to [the documentation](http://fkie.github.io/catkin_lint/messages/), that happens when `You have listed an include path in the INCLUDE_DIRS stanza of the catkin_package() command, but that path is not mentioned in any include_directories() call.`

The problem is that it is not used in any `include_directories call`, its used in a `target_include_directories` call of a library target within the package. I have a library in the package that I am exposing to other packages by exporting the library through `catkin_package()`. Without the INCLUDE_DIRS stanza listing the library's include path, the other package cannot access the library's header.

roehling, 2020-08-31:
Support for `target_include_directories()` has been added in version 1.6.7 (March 2020). Are you using the latest version?

gavanderhoorn, 2020-08-31:
Probably related: ros-industrial/industrial_ci#554.

wmmc88, 2020-08-31:
I updated to using the ppa and issue was resolved.

closed 2020-08-31 by wmmc88