fkie/catkin_lint #89: catkin_lint does not support package format3 condition tag

fmessmer, 2020-10-21:
in preparation to support noetic, I recently added noetic jobs to one of the repos I maintain: see https://github.com/ipa320/cob_control/pull/243
there we use `industrial_ci` for CI and we have `catkin_lint` activated

as stated in the [Noetic Migration Guide](http://wiki.ros.org/noetic/Migration#Use_Orocos_KDL_and_BFL_rosdep_keys), I changed the rosdep key for KDL to the following in order to make our `melodic-devel` branch compatible with both distros `melodic` and `noetic`:
```
  <depend condition="$ROS_DISTRO == melodic">orocos_kdl</depend>
  <depend condition="$ROS_DISTRO == noetic">liborocos-kdl-dev</depend>
  <depend condition="$ROS_DISTRO == noetic">liborocos-kdl</depend>
```

however, the `catkin_lint` step reports the following error now for the `noetic` job:
```
Starting function 'catkin_lint'
catkin_lint: downloading noetic package index from https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
catkin_lint: downloading package manifest for 'fcl'
cob_frame_tracker: package.xml: error: unknown package 'orocos_kdl'
     * You are referring to a package which seems to be neither a
     * catkin package nor a known system dependency. You may have
     * misspelled the name, or your rosdep database needs to be
     * refreshed with "rosdep update".
     * You can ignore this problem with --ignore unknown_package
cob_model_identifier: package.xml: error: unknown package 'orocos_kdl'
cob_obstacle_distance: package.xml: error: unknown package 'orocos_kdl'
cob_twist_controller: package.xml: error: unknown package 'orocos_kdl'
catkin_lint: checked 17 packages and found 4 problems
catkin_lint: 11 messages have been ignored. Use --show-ignored to see them
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
Function 'catkin_lint' returned with code '1' after 0 min 2 sec
The command ".industrial_ci/travis.sh" exited with 1.
```
which makes me believe that `catkin_lint does not evaluate the `condition argument in the `depend` - still thinking `orocos_kdl` is specified as a dependency to look up in `rosdep`

roehling, 2020-10-21:
Which version are you using? Support for package format 3 has been added in version 1.6.9

gavanderhoorn, 2020-10-21:
Perhaps https://github.com/ros-industrial/industrial_ci/issues/554?

fmessmer, 2020-10-21:
thanks!
focal seems to use `1.6.6`: https://packages.ubuntu.com/focal/catkin-lint
obviously I'll have to update the `industrial_ci` branch to include https://github.com/ros-industrial/industrial_ci/pull/555 - which pulls in latest `catkin-lint` from pip

closed 2020-10-21 by fmessmer

gavanderhoorn, 2020-10-21:
Actually, with ros-infrastructure/reprepro-updater#81 merged, even that should not be necessary any more.