fkie/catkin_lint #90: catkin_lint does not handle LANGS in generate_messages

bajo, 2020-10-21:
Setting LANGS in generate_messages allows us to only generate files for a specific language (eg. cpp, or python). https://docs.ros.org/en/api/genmsg/html/usermacros.html

catkin_lint treats LANGS however as a dependency, thus, generating these errors on a private package.

```
error: message dependency 'LANGS' is not listed in catkin_package()
error: message dependency 'LANGS' is not listed as build_depend
error: message dependency 'LANGS' is not listed as build_export_depend
error: unconfigured message dependency 'LANGS'
```

roehling, 2020-10-21:
Are you using the outdated 1.5.2 version from the ROS package archive? The good news is, that version version is just [about to be updated](https://github.com/ros-infrastructure/reprepro-updater/pull/81).

bajo, 2020-10-21:
> Are you using the outdated 1.5.2 version from the ROS package archive? The good news is, that version version is just [about to be updated](https://github.com/ros-infrastructure/reprepro-updater/pull/81).

Yes, I'm using version 1.5.2. 
Good to know that this issue will be fixed soon. Thank for the quick reply.

closed 2020-10-21 by bajo