fkie/catkin_lint #91: Package path name check doesn't take into account symlinks

Timple, 2020-11-05:
We tend to symlink packages to our workspace. Because of this the absolute path name might not be equal to the package name. 

**background:**
Using only a single package from a multi-package repository leads to compiling a lot of packages for nothing. Therefore we check-out these repositories to a different location and symlink the packages to our workspace.
For consistency we also do this for single-package repositories.

This all works fine but catkin_lint fails to detect the name of the symlink but takes the name of the absolute path (which we postpend with `.git`.

**question:**
However the symlink name in the workspace is correct. Could the actual path as seen by ROS be taken into account?

roehling, 2020-11-05:
I thought I dealt with this issue for you already (#64). Is this still broken, are do you just happen to use an outdated version? It should be fixed since version 1.6.1

Timple, 2020-11-05:
Hmmm, you make an excellent point there!
Our CI suddenly failed on this, so let me figure out the versions first!

Timple, 2020-11-05:
On my computer I have the following:

```
$ ls -lah beetle
lrwxrwxrwx 1 tim tim 68 Nov  5 12:44 beetle -> /long/path/to/beetle.git/.

$ catkin_lint beetle -W2
catkin_lint: checked 1 packages and found 0 problems

$ cd beetle && catkin_lint -W2
beetle: notice: package path name '/long/path/to/beetle.git' differs from package name
catkin_lint: checked 1 packages and found 1 problems
```

Version:
```
$ catkin_lint --version
1.6.10

$ apt-cache policy python-catkin-lint
python-catkin-lint:
  Installed: 1.6.10-1~18.04
  Candidate: 1.6.10-1~18.04
  Version table:
 *** 1.6.10-1~18.04 500
        500 http://packages.ros.org/ros-testing/ubuntu bionic/main amd64 Packages
        500 http://packages.ros.org/ros-testing/ubuntu bionic/main i386 Packages
        500 http://packages.ros.org/ros2/ubuntu bionic/main amd64 Packages
        100 /var/lib/dpkg/status

```

Note that it doesn't seem like I'm using the recommended ppa. But it's version `1.6.10` nonetheless.

roehling, 2020-11-05:
Alright. Can you give my the actual folder tree structure, so I can reproduce this locally? I don't need the actual data, only which directories exist and where the symlinks go, and then I'll mock a package myself.

roehling, 2020-11-05:
Actually, I don't need this. The problem is `os.getcwd()` which will always return a fully resolved path, so the symlink is lost.

roehling, 2020-11-05:
As it turns out, there is no easy fix for this, because the notion of current directory is linked to the real path and not a symlink which happens to point there. For usability reasons, many shells (but not all) keep track of the logical path that was used to reach a directory and put it in the `PWD` environment variable. However, this is purely informational and will not automatically update when a subprocess (such as Python) changes the working directory with `os.chdir`.

So I settled for the following strategy: `catkin_lint` will look at `PWD`, and if it set and points to the same physical location as `os.getcwd()`, it will be used as working directory.

closed 2020-11-05 by roehling

Timple, 2020-11-05:
Awesome, thanks for the quick fix!

commit 2f44a57f (2020-11-05): Keep symlinks in working directory path if possible

Fixes #91