fkie/catkin_lint #92: include directory check

asherikov, 2020-11-06:
Hi, nice tool, thanks.

I wonder if it is possible to add a warning if headers installed to CATKIN_PACKAGE_INCLUDE_DESTINATION don't match
'include/<package_name>/*' pattern as in http://wiki.ros.org/catkin/CMakeLists.txt#Installing_header_files.

Also, personally I would prefer to get relative paths to offending files instead of '<package_name>: package.xml ...', it would save me a couple of seconds when fixing an issue ;).

closed 2020-11-10 by roehling

roehling, 2020-11-10:
I added a check for the `CATKIN_PACKAGE_INCLUDE_DESTINATION`.

As for the paths, they are relative to the package directory already. All you need to do is run `roscd $PACKAGE` first. Revamping my output code (that supports different output formats) would be more work than you'll ever save, so I'll have to decline that request.

asherikov, 2020-11-10:
thanks for a quick response

commit 2d78d09a (2020-11-10): Check if headers are installed to the package-specific include path

Fixes #92