fkie/catkin_lint #96: Best practices for ExternalProject

stertingen, 2021-02-22:
Hello folks.

Currently I'm trying to integrate an [external C++ library](https://github.com/CopernicaMarketingSoftware/AMQP-CPP) into my ROS workspace using [ExternalProject](https://cmake.org/cmake/help/v3.5/module/ExternalProject.html). I cannot use FetchContent which is not available on Kinetic and Melodic.

While it is possible to perform this task working for both devel and install destinations, catkin_lint has a lot to complain about:

```cmake
cmake_minimum_required(VERSION 3.5)
project(amqpcpp)

find_package(catkin REQUIRED)
find_package(OpenSSL REQUIRED)

# Fetch version from package.xml
catkin_package_xml()

# Download AMQP-CPP and install into devel space
include(ExternalProject)
externalproject_add(${PROJECT_NAME}_ep
    # Downloading archive is faster and less verbose than using git
    URL "https://github.com/CopernicaMarketingSoftware/AMQP-CPP/archive/v${${PROJECT_NAME}_VERSION}.tar.gz"
    # Forward install prefix to external project
    CMAKE_ARGS
        -DCMAKE_INSTALL_PREFIX=<INSTALL_DIR>
    # Configuration of AMQP-CPP
    CMAKE_CACHE_ARGS
        -DAMQP-CPP_BUILD_EXAMPLES:BOOL=OFF
        -DAMQP-CPP_BUILD_SHARED:BOOL=ON
        -DAMQP-CPP_LINUX_TCP:BOOL=ON
    # Output files of external project
    BUILD_BYPRODUCTS
        <INSTALLL_DIR>/lib/lib${PROJECT_NAME}.so
        <INSTALLL_DIR>/lib/lib${PROJECT_NAME}.so.4.3
    # Set devel space as install prefix
    INSTALL_DIR "${CATKIN_DEVEL_PREFIX}"
)

# Workaround for cmake complaining about missing include directory
file(MAKE_DIRECTORY ${CATKIN_DEVEL_PREFIX}/include)

# Workaround for catkin_lint complaining about external include directory
list(APPEND ${PROJECT_NAME}_INCLUDE_DIRS ${CATKIN_DEVEL_PREFIX}/include)

# Declare libraries and dependencies for catkin
# Include directories are declared above.
catkin_package(
    LIBRARIES ${PROJECT_NAME}
    DEPENDS OPENSSL
)

# Add interface library which can be used for non-install builds
add_library(${PROJECT_NAME} INTERFACE)
# Declare include directories for our interface target (from installed AMQP-CPP)
target_include_directories(${PROJECT_NAME} SYSTEM INTERFACE ${CATKIN_DEVEL_PREFIX}/include)
# Declare libraries for our interface target (from installed AMQP-CPP)
target_link_libraries(${PROJECT_NAME} INTERFACE ${CATKIN_DEVEL_PREFIX}/lib/lib${PROJECT_NAME}.so OpenSSL::SSL)
# Make our interface target depend on the external project
add_dependencies(${PROJECT_NAME} ${PROJECT_NAME}_ep)

# Workaround for catkin_lint complaining about amqpcpp not being installed
install(TARGETS ${PROJECT_NAME}
    DESTINATION ${CATKIN_PACKAGE_LIB_DESTINATION}
)

# Install library binaries (from AMQP-CPP which is currently installed to devel space)
install(FILES ${CATKIN_DEVEL_PREFIX}/lib/lib${PROJECT_NAME}.so ${CATKIN_DEVEL_PREFIX}/lib/lib${PROJECT_NAME}.so.4.3
    DESTINATION ${CATKIN_PACKAGE_LIB_DESTINATION}
)
# Install header files (from AMQP-CPP which is currently installed to devel space)
install(FILES ${CATKIN_DEVEL_PREFIX}/include/${PROJECT_NAME}.h
    DESTINATION ${CATKIN_GLOBAL_INCLUDE_DESTINATION}
)
install(DIRECTORY ${CATKIN_DEVEL_PREFIX}/include/${PROJECT_NAME}/
    DESTINATION ${CATKIN_PACKAGE_INCLUDE_DESTINATION}
    FILES_MATCHING PATTERN "*.h"
)
```

Especially the lines
```cmake
list(APPEND ${PROJECT_NAME}_INCLUDE_DIRS ${CATKIN_DEVEL_PREFIX}/include)
```
and
```cmake
install(TARGETS ${PROJECT_NAME}
    DESTINATION ${CATKIN_PACKAGE_LIB_DESTINATION}
)
```
bother me because they clearly circumvent checks performed by catkin_lint.

Even with these workarounds, I get warnings about installing files from the devel space (because that's where the external project is installed to). I think most issues boil down to doing things in the devel space and exporting references to the devel space.

So can I somehow use ExternalProject with catkin and still satisfy catkin_lint?

gavanderhoorn, 2021-02-22:
I'm not the maintainer here, nor is this an answer, but do you really need to wrap your library in a Catkin package?

If you're just trying to use `amqpcpp` as a system dependency, it would not be needed (as it would just be another system dependency, found using `find_package(..)`).

If you're looking to be able to build it as part of a Catkin workspace, it would not be needed (as you could just place a minimal `package.xml` in your projects directory, and set the `build_type` to `cmake`).

If you're attempting to release `amqpcpp` as a package on the ROS buildfarm -- or a private buildfarm -- you could use a [thirdparty release](http://wiki.ros.org/bloom/Tutorials/ReleaseThirdParty).

stertingen, 2021-02-22:
> 
> 
> I'm not the maintainer here, nor is this an answer, but do you really need to wrap your library in a Catkin package?
> 
> If you're just trying to use `amqpcpp` as a system dependency, it would not be needed (as it would just be another system dependency, found using `find_package(..)`).
> 
> If you're looking to be able to build it as part of a Catkin workspace, it would not be needed (as you could just place a minimal `package.xml` in your projects directory, and set the `build_type` to `cmake`).
> 
> If you're attempting to release `amqpcpp` as a package on the ROS buildfarm -- or a private buildfarm -- you could use a [thirdparty release](http://wiki.ros.org/bloom/Tutorials/ReleaseThirdParty).
> 
> Edit: and this is highly opinionated, but:
> 
> > I cannot use FetchContent which is not available on Kinetic and Melodic.
> 
> Both `FetchContent(..)` and `ExternalProject(..)` are really not very nice: they mix the responsibility of building a piece of software with making sure the dependencies are present. There is a very clear separation of concerns in a typical ROS workflow, and `FetchContent(..)` and `ExternalProject(..)` break it.

I totally aggree with you that including amqpcpp as system package would be the better solution. However, creating and maintaining a release repository seems to require some amount of extra maintenance effort. Just throwing a package.xml into the project does not work for catkin_make.

Also, integrating own modifications (using a forked version for development purposes) does not seem to be possible while I could always change the URL in the CMakeLists.txt.

gavanderhoorn, 2021-02-22:
> However, creating and maintaining a release repository seems to require some amount of extra maintenance effort. 

more than what you're doing now? It's a trade-off, but a third-party release is essentially a one-time configuration job, followed by periodic rebasing of your manifest + any potential patches you want to include + running `bloom` again.

That's not really that much work.

> Just throwing a package.xml into the project does not work for catkin_make.

No, as it does not support mixed-build-tool builds.

You'd use `catkin_tools` or `catkin_make_isolated` for those cases.

> Also, integrating own modifications (using a forked version for development purposes) does not seem to be possible while I could always change the URL in the CMakeLists.txt.

I'm not sure I understand this.

roehling, 2021-02-22:
I don't think there's much to add on my part. @gavanderhoorn pretty much laid out the most sensible options and I don't think there's enough of a use case for me to add support for `ExternalProject` to catkin_lint. I can turn this into a discussion if you like, as this might be off-topic, but it still provides valuable information which others might find useful.

stertingen, 2021-02-22:
> > However, creating and maintaining a release repository seems to require some amount of extra maintenance effort.
> 
> more than what you're doing now? It's a trade-off, but a third-party release is essentially a one-time configuration job, followed by periodic rebasing of your manifest + any potential patches you want to include + running `bloom` again (that is: _if_ you want to release updates).
> 
> That's not really that much work, and seems comparable to what you'd be doing with the wrapper you're creating now.

Once the package is released into the ROS ecosystem, people might depend on it even when my current project does not use this library any more. So releasing something with bloom comes with a responsibility to maintain the package, I think.

> > Just throwing a package.xml into the project does not work for catkin_make.
> 
> No, as it does not support mixed-build-tool builds.
> 
> You'd use `catkin_tools` or `catkin_make_isolated` for those cases.

Yes. I do see the rationale for using these, but for deveopment I prefer catkin_make due to the faster build times.

> > Also, integrating own modifications (using a forked version for development purposes) does not seem to be possible while I could always change the URL in the CMakeLists.txt.
> 
> I'm not sure I understand this.

There are situations in which I need a modified version of a package (for example when working on a PR) while testing my ROS workspace. In such situations, I just put the source package into the workspace and it gets built by catkin automatically.

> I don't think there's much to add on my part. @gavanderhoorn pretty much laid out the most sensible options and I don't think there's enough of a use case for me to add support for `ExternalProject` [...]

Yes, this can be turned into a discussion. Sorry for opening an issue for this!

gavanderhoorn, 2021-02-22:
final comment:

> There are situations in which I need a modified version of a package (for example when working on a PR) while testing my ROS workspace. In such situations, I just put the source package into the workspace and it gets built by catkin automatically.

I don't really understand how having a `package.xml` in an otherwise plain CMake project hinders this sort of workflow.

Google Cartographer for instance is setup and distributed in exactly the same way, and has almost infinite forks which can all be built from source in a Catkin workspace.

Perhaps unclear, but adding a manifest does not equal 3rd party release. It's perfectly possible to have a manifest in an unreleased package.

roehling, 2021-02-23:
> Yes, this can be turned into a discussion. Sorry for opening an issue for this!

No need to be sorry. It's a legitimate question to ask, and I hope the answers have been helpful for you.

closed 2021-02-23 by roehling