fkie/catkin_lint #98: ignore_once for launch files?

gavanderhoorn, 2021-06-23:
A quick test made it seem that's not supported, but I didn't really try very hard:

```xml
  <!-- catkin_lint: ignore_once launch_depend -->
  <node name="rviz" pkg="rviz" type="rviz" args="..." />
```

This is a package where the `.launch` file is basically "extra", and adding a strict dependency on `rviz` would complicate installations on headless systems.

gavanderhoorn, 2021-06-23:
I'm guessing something around here needs to change, but I'm not yet sure what:

https://github.com/fkie/catkin_lint/blob/6a3c610e1804b4c2dd6b936535fce9999900905b/src/catkin_lint/checks/manifest.py#L97-L109

Timple, 2021-06-24:
We typically put these launch files in the `test` folder. And add a `test_depend` on `rviz`.
But that still feels a bit like a workaround since it doesn't really test anything. So this could be nice.

gavanderhoorn, 2021-06-24:
> We typically put these launch files in the `test` folder. And add a `test_depend` on `rviz`.

yeah, that's not really what we're going for.

it'd be nice if the ROS manifest format supported something like optional dependencies, which would be ideal for this.

Another option would be using a different package to host such optional `.launch` files, but that's not doable in this case.

roehling, 2021-06-24:
Would it be feasible for you to add `example` to the file names or put them in a folder where the name contains `example`? Because there is an exception for binaries with `test` or `example` in their names that `catkin_lint` will not complain if they are not installed, and it would be nicely consistent to have the same convention for launch files, too.

gavanderhoorn, 2021-06-24:
They're not actually either tests or examples. They're just regular `.launch` files, which happen to have dependencies we can't really hard-code (the headless example in the OP is a good one actually).

roehling, 2021-06-24:
Well, as it turns out, that was easier to implement than I tought initially.

closed 2021-06-24 by roehling

gavanderhoorn, 2021-06-24:
Awesome, thanks a lot :+1:

gavanderhoorn, 2021-06-24:
And I almost feel bad for asking, but will you be cutting a new release as well with this included?

roehling, 2021-06-24:
I was going to wait for another day or two, as experience tells me issues come in clusters. Then again, there are plenty of natural numbers available, I'm not going to run out for patch releases. 😉

commit 38f67cfb (2021-06-24): Allow {ignore,ignore_once,report} pragmas for launch_depend in launch files

Fixes #98
commit cebc8254 (2021-06-24): Improve ignore logic for launch files

Fixes #98 again. Better this time.