fkie/catkin_lint #102: cannot lint foo: list index out of range

lucasw, 2021-09-20:
catkin_lint  with --debug:

```
Traceback (most recent call last):
  File "/usr/bin/catkin_lint", line 51, in <module>
    main()
  File "/usr/share/catkin-lint/catkin_lint/main.py", line 320, in main
    sys.exit(run_linter(args))
  File "/usr/share/catkin-lint/catkin_lint/main.py", line 280, in run_linter
    linter.lint(path, manifest, config=config)
  File "/usr/share/catkin-lint/catkin_lint/linter.py", line 709, in lint
    cb(info)
  File "/usr/share/catkin-lint/catkin_lint/checks/manifest.py", line 99, in on_final
    if args[0] == "catkin_lint:" and args[1] in ["ignore_once", "ignore", "report"]:
IndexError: list index out of range
```

I could have added something odd to the package that is causing this to happen, though rolling back doesn't help.

I can probably trace this back but help would be appreciated, I see args above come out of node.text `                                args = (node.text or "").split()`, so far I see that comes from `Message`, which comes from `report()`.

gavanderhoorn, 2021-09-20:
Do you have a comment in your manifest like this:

```xml
<!-- catkin_lint: <some other string> -->
```

If so, it's probably triggering the parser making it believe you're trying to configure an ignore.

lucasw, 2021-09-20:
I bisect down to the commit that causes the problem, and I did change a launch files- I had a commented out block with nothing in it, I can remove it and the issue goes away:

```
  <!--
  -->
```

I can put a single character into it like ``, then it is fine.

commit ffb46f42 (2021-09-20): Handle empty XML comments

Fixes #102
closed 2021-09-20 by roehling