fkie/catkin_lint #104: [feature] Support junit/xunit XML output

MatthijsBurgh, 2022-04-02:
I would like to integrate this as a catkin run test. Though this requires xunit/junit xml output. Which isn't supported now.

roehling, 2022-04-05:
I don't think the xunit XML format is a very good fit for `catkin_lint`: you can only indicate Pass/Fail, which is fine for unit tests but awfully limited for a linter. Besides, `catkin_lint` has a well-defined XML output format already, which you can transform to anything you like using XSLT. Alternatively, you can wrap the `catkin_lint` text output in some XML boilerplate to create your own xunit result file, such as

```
import sys
import subprocess

result = subprocess.run(["catkin_lint"] + sys.argv[1:], capture_output=True, text=True)
if result.returncode == 0:
    sys.stdout.write('<testcase id="catkin_lint"/>\n')
else:
    sys.stdout.write('<testcase id="catkin_lint"><failure><![CDATA[')
    sys.stdout.write(result.stdout)
    sys.stdout.write(result.stderr)
    sys.stdout.write(']]></failure></testcase>\n')
```

MatthijsBurgh, 2022-04-05:
@roehling thanks for the example code