fkie/catkin_lint #105: Library install location

MatthijsBurgh, 2022-04-15:
The library is not installed in a common location for python libraries. The library is installed in `/usr/share` (See end of issue for complete file list). (I have tested the focal, ros servers and your personal PPA. All install in the same  location.) While python only looks in the following directories by default:

```py
print(sys.path)
['/opt/ros/noetic/lib/python3/dist-packages',  # Only when you have ros
 '/usr/lib/python38.zip',
 '/usr/lib/python3.8',
 '/usr/lib/python3.8/lib-dynload',
 '/home/USER/.local/lib/python3.8/site-packages',
 '/usr/lib/python3/dist-packages',
 '/usr/local/lib/python3.8/dist-packages',
 '/usr/lib/python3.8/dist-packages']
```
The pip version does install in the local site-packages folder (`/home/USER/.local/lib/python3.8/site-packages`)

I know I can manually add directories so that python will be able to import the library, but the current situation isn't very useful. So it would be very helpful when you would change the install location.

----

Complete package file list:

```bash
$ dpkg -L catkin-lint
/.
/usr
/usr/bin
/usr/share
/usr/share/bash-completion
/usr/share/bash-completion/completions
/usr/share/bash-completion/completions/catkin_lint
/usr/share/catkin-lint
/usr/share/catkin-lint/catkin_lint
/usr/share/catkin-lint/catkin_lint/__init__.py
/usr/share/catkin-lint/catkin_lint/__main__.py
/usr/share/catkin-lint/catkin_lint/_version.py
/usr/share/catkin-lint/catkin_lint/checks
/usr/share/catkin-lint/catkin_lint/checks/__init__.py
/usr/share/catkin-lint/catkin_lint/checks/build.py
/usr/share/catkin-lint/catkin_lint/checks/cuda.py
/usr/share/catkin-lint/catkin_lint/checks/manifest.py
/usr/share/catkin-lint/catkin_lint/checks/misc.py
/usr/share/catkin-lint/catkin_lint/checks/python.py
/usr/share/catkin-lint/catkin_lint/cmake.py
/usr/share/catkin-lint/catkin_lint/diagnostics.py
/usr/share/catkin-lint/catkin_lint/environment.py
/usr/share/catkin-lint/catkin_lint/linter.py
/usr/share/catkin-lint/catkin_lint/main.py
/usr/share/catkin-lint/catkin_lint/output.py
/usr/share/catkin-lint/catkin_lint/ros.py
/usr/share/catkin-lint/catkin_lint/util.py
/usr/share/catkin-lint/catkin_lint-1.6.17.egg-info
/usr/share/catkin-lint/catkin_lint-1.6.17.egg-info/PKG-INFO
/usr/share/catkin-lint/catkin_lint-1.6.17.egg-info/dependency_links.txt
/usr/share/catkin-lint/catkin_lint-1.6.17.egg-info/entry_points.txt
/usr/share/catkin-lint/catkin_lint-1.6.17.egg-info/requires.txt
/usr/share/catkin-lint/catkin_lint-1.6.17.egg-info/top_level.txt
/usr/share/catkin-lint/run
/usr/share/doc
/usr/share/doc/catkin-lint
/usr/share/doc/catkin-lint/changelog.Debian.gz
/usr/share/doc/catkin-lint/copyright
/usr/share/man
/usr/share/man/man1
/usr/share/man/man1/catkin_lint.1.gz
/usr/share/python3
/usr/share/python3/runtime.d
/usr/share/python3/runtime.d/catkin-lint.rtupdate
/usr/bin/catkin_lint
```

roehling, 2022-04-15:
The install location is intentional, because the module is a private implementation detail and you are not supposed to import `catkin_lint`. In particular, I don't make any effort to keep the code backwards compatible (although it is rather stable in practise these days).

MatthijsBurgh, 2022-04-15:
I am not really interested in the actual library. I just wanted to import `prepare_arguments` and `run_linter`. So I can just run `catkin_lint` inside my process, instead of using subprocess.

closed 2022-05-06 by MatthijsBurgh