fkie/catkin_lint #107: Option to ignore/disable directories

MatthijsBurgh, 2022-10-03:
Is there an option or would it be possible to add an option to ignore/disable directories?

As I have multiple packages which include an 3rdparty library. Often these libraries include executable files, which aren't installed. I don't want to disable that issue, as I want it to be checked for my package, just not for the 3rdparty library.

An option to ignore directories(and maybe individual files) would be very useful for this.

I would be open to do most of the work for such a feature as you could point me where to go.

roehling, 2022-10-03:
I've added some logic to skip this check for sub projects (which are skipped during linting anyway). If that is not sufficient, a very simple solution is to add a `CATKIN_IGNORE` marker file to the third-party subdirectories.

MatthijsBurgh, 2022-10-04:
Thanks. I already figured out that a `CATKIN_IGNORE` would do the job. Maybe this could be documented.

commit 58f0298f (2023-01-03): Document CATKIN_IGNORE

Closes #107
closed 2023-01-03 by roehling